talkkonnect/talkkonnect #79: Cmd pause or stop/start Talkkonnect

f59465-Clem, 2023-03-16:
Hello everyone,

I take control of my Rpi several kilometers away via SSH, how can I pause or stop Talkkonnect and thus start it again via a command for example?

Unless Talkkonnect already does it by default...

When I do ctrl-c, it restarts after 30 seconds.

Cordially

Propeller-v8, 2023-03-16:
Hi Clem,

"systemctl stop talkkonnect" and "systemctl start talkkonnect" should work.

Greets
Andre

f59465-Clem, 2023-03-16:
Hi Andre, 

Thanks it works, it's efficient. 

An idea for future development, a command that handles the <commentbuttonpin> function to start or stop TK.

Cordially

talkkonnect, 2023-03-16:
Clem,
What is your use case? talkkonnect is an application should be trivial to start and stop by using OS commands like other commands. I need to understand what you are trying to do exactly.

f59465-Clem, 2023-03-16:
Hi Suvir,

Using RF/voip gateway, I was looking to pause TK (without disconnecting from Mumble/then start) remotely via a command with message display in Mumble, in the event of a problem or abuse of the RF/voip system.

Maybe an idea to add functionality.

Cordially

talkkonnect, 2023-03-16:
By stopping do you want talkkonnect to keep running but do something like disable GPIO pin remotely so that the rx or tx functionality of talkkonnect stops working? If that is the case then can be done. Send command to talkkonnect via http api and mqtt?

suvir

f59465-Clem, 2023-03-16:
_By stopping do you want talkkonnect to keep running but do something like disable GPIO pin remotely so that the rx or tx functionality of talkkonnect stops working?_ Yes exactly.
 _Send command to talkkonnect via http api and mqtt?_Yes or ssh cmd.

talkkonnect, 2023-03-21:
Understood and I can add this functionality for the console to be able to disable and enable GPIO for RX and TX as a command key. It will take some time and I have put it on the todo list for talkkonnect.

thanks for the suggestion.
suvir

closed 2023-03-25 by talkkonnect