talkkonnect/talkkonnect #81: GPIO LEDs not working - Version 2

mudislandkid, 2023-04-04:
Hi There,

Firstly, thank you so much for putting this together! It's amazing.

I'm having some issues with getting certain LEDs to work. Online and TX work fine. However I cannot get the participants, heartbeat or voice activity lights to work. There seems to be quite a difference between version 1 and 2 of the xml files. v1 explicitly calls out each GPIO where as v2 doesn't. Can you point me in the right direction for how to enable/disable these? Similar with the TX Toggle and attention light etc. I would like to be able to access all of these via the GPIO.

I'm running on a RPI3B.

Many Thanks

talkkonnect, 2023-04-04:
Hi,
Thank you for your interest in talkkonnect. Can you please let me know what exact version of talkkonnect you are running. If you are using the version I recently put up for raspberry pi 3 all GPIO should work as expected. On the older images of version 2 there were some bugs.

I suggest you completely replace your xml config file with the version 2 config found in the sample-config folder. Start modifying using the usb example https://github.com/talkkonnect/talkkonnect/blob/main/sample-configs/talkkonnect-version2-usb-gpio-example.xml

to get heartbeat to work 

<pin direction="output" device="led/relay"     name="heartbeat" pinno="2" type="gpio" chipid="0" inverted="false" enabled="true"/>

this means you are using GPIO 2 physical pin 3 as example above. The name is heartbeat so that means it will map the heartbeat to this GPIO pin. Chip Id 0 means you are using direct GPIO of raspberry pi and not an i2c expander chip. Inverted false means that use positive logic and not inverter logic, here it doesn’t really matter for heartbeat but matters for when connecting for example to a radio for the voiceactivity function. Enabled has to be true.

next for heartbeat you need also to enable this section and make changes to the timing

<heartbeat enabled="true">
        <heartbeatledpin/>
        <periodmsecs>2000</periodmsecs>
        <ledonmsecs>1000</ledonmsecs>
        <ledoffmsecs>1010</ledoffmsecs>
      </heartbeat>

let me know how you get along.

Suvir

mudislandkid, 2023-04-05:
Thank you for the quick response. It seems using a fresh xml config worked. I also misunderstood the IO section as I thought that was only enabled if I enabled the GPIO expansion. So that now makes sense as I was struggling to figure out where you were setting the pins and so was using the old layout from a v1 config. Heartbeat, participants and voice activity etc all work now thank you!

On another note, when it comes to the TTS, I'm only getting some of them working. Talkkonnect loaded works as does the message "You are currently alone in the channel root" when I first join the server. However when I change channels, or when someone enters or leaves the channel and I am left 'alone'. I see the message displayed on the terminal output, but there is no TTS. If you see the below output...the first text= plays audio but very quietly....the second and third do not. Where do I set the volume for these? Also, any idea why the 2nd and third messages do not produce audio?

Many Thanks

```
[ alert ] 2023/04/05 09:30:51 htgotts.go:41: text=You are Currently Alone in The Channel Root, destination=local playBackVolume=1 duration=0 loop=1 language=en
[  info ] 2023/04/05 09:30:56 clientcommands.go:318: Channel  Root  has no other participants                               
[  info ] 2023/04/05 09:30:56 clientcommands.go:234: Joined Channel Name:  Theatre Comms  ID  1                             
[ alert ] 2023/04/05 09:30:56 htgotts.go:41: text=There Are Currently 2 Users in The Channel Theatre Comms, destination=local playBackVolume=1 duration=0 loop=1 language=en
[  info ] 2023/04/05 09:30:56 client.go:444: Client Is Already Registered                                                   
[  info ] 2023/04/05 09:30:58 onevent.go:251: User pican1s changed channel                                                  
[  info ] 2023/04/05 09:31:28 onevent.go:251: User gregs changed channel                                                    
[ alert ] 2023/04/05 09:31:30 htgotts.go:41: text=There Are Currently 2 Users in The Channel Theatre Comms, destination=local playBackVolume=1 duration=0 loop=1 language=en
[  info ] 2023/04/05 09:31:32 onevent.go:251: User gregs changed channel  
```

talkkonnect, 2023-04-05:
volume = 1 is volume i set to 1% try to increase the volume for volume=100 for all items first then adjust them down accordingly. First start at volume=100 for all items that are enabled=true

mudislandkid, 2023-04-05:
I've tried that. There was no sound entry that had a volume of 1 to begin with. The TTS message is still showing 1 in the volume the first time it is played which I can just about hear. However when it is triggered again, there is no sound at all. There is also no sound being triggered when I quit talkkonnect, even though that is enabled. - Sorry about this! It's probably something silly that I've missed! Should I open a different issue for this?

talkkonnect, 2023-04-05:
OK I checked and looked at the source code, its a bug for some reason I have set all subsequent tts volumes hardcoded as 1, should be 100 in the source code. Looking at the code from github this is my analysis. I have to actually test this at home with my talkkonnect build to verify and fix this. 

Thank you for the bug report. I made the tts for those people who are working and cannot take their eyes off their work, also for one of my good friends who is blind. However it is was not well tested for every message.

mudislandkid, 2023-04-05:
No worries at all! What you have built is amazing! So I really appreciate you helping on this. My plan is to try and use this for wireless comms in a school theatre. The TTS will be useful as will help people be aware when someone joins or leaves the channel. Will also be looking into how I can utilise the attention LED a bit easier. Ideally having a button that will light up the attention LED's on all other connected clients. This is useful in a theatre environment where someone may have their headset off and we require them to put them back on and listen! Thanks so much again.

talkkonnect, 2023-04-05:
There is an attention message that you can blink an led to get someones attention
<attentionblinktimes>20<attentionblinktimes>
<attentionblinkmsecs>300<attentionblinkmsecs>
you can send the attention by httpai or mqtt

talkkonnect, 2023-04-06:
I have fixed the TTS issue for volume and participant count in channel announcement. Fixes released in
version 2.20.01 released Apr 6  2023.

closed 2023-04-06 by talkkonnect