talkkonnect/talkkonnect #82: Audio Card Issue

mudislandkid, 2023-04-06:
Thanks for fixing the TTS volume issues. I've now updated to the latest version.

However now I have a new issue. When I load talkkonnect I get this error:

`failed to execute "pactl list sinks" (exit status 1)`

I can hear sound and I can speak just fine. The alsamixer shows my USB Headset as the default audio source. I have not had this error before. Previously when audio wasn't setup properly I'd get a 'failed to unmute x device', but now I get the above as well as it saying it cannot get volume data from the sound card.

Running on RPI 3B with Bullseye. 
Generic USB Sound card (was working before)
This is set as card 1 as card 0 is the HDMI output.
Installed talkkonnect manually following the readme instructions.
Running latest version of Go. go1.20.3

talkkonnect, 2023-04-07:
Should run raspberry command line version without GUI and without pulse audio installed.

talkkonnect, 2023-04-08:
Since this is sound card config issue in OS allow me to close this ticket.

closed 2023-04-08 by talkkonnect

Keksstar, 2024-02-25:
> Thanks for fixing the TTS volume issues. I've now updated to the latest version.
>
> However now I have a new issue. When I load talkkonnect I get this error:
>
> `failed to execute "pactl list sinks" [...]

Have you solves your issue? I have the same. -.-