talkkonnect/talkkonnect #83: Announcement played

f59465-Clem, 2023-04-07:
Hello,
 Using talkkonnect as an RF/VoIP gateway.

I would like to broadcast an announcement (.wav file, active gateway relay) about 10s after the connection of Talkkonnect to the server, or if not possible via a gpio command switch.

I would like this audio announcement to be played only on the RF side, not Voip, just to inform users that the radio relay is active.
Is this possible?

Thanks
Clem

talkkonnect, 2023-04-08:
Yes this should be possible with http api. But there is nothing like after connecting 10 seconds then play. It is adhoc by httpapi only currently.

f59465-Clem, 2023-04-08:
ok Suvir,
I will study http api, and make a bash script with the command.
I would return.

Thank you for the new Features, I have not yet understood how to update without destroying my .img
73'

Propeller-v8, 2023-04-08:
Clem,

update routine:
To update to the lastest version release of talkkonnect in the image you can:
cd to the root directory

> rm update-talkkonnect.sh*
> wget https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/update-talkkonnect.sh
> chmod +x ./update-talkkonnect.sh
> /root/update-talkkonnect.sh

Easy
greets Andre

f59465-Clem, 2023-04-08:
404 not found...

talkkonnect, 2023-04-09:
Clem,

have a look here https://github.com/talkkonnect/talkkonnect/tree/main/scripts
choose the update script that matches your use case. See the original image if 32 or 64 bit os.
Click on the image then click on raw in your browser, once that is done copy the raw url of the desired script.
Then use Andre Steps as above changing the URL to the appropriate one you just selected.

I have updated the update scripts to update golang also as the new versions need the latest version of golang.
Using the new correct scripts the build will not break the upgrade and talkkonnect should work as expected after
upgrading.

f59465-Clem, 2023-04-09:
Suvir,

In the meantime I have adapted the command according to my image (PiZ2x, 64 bits, USB sound card):

rm update-talkkonnect.sh*
wget https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/update-talkkonnect-64bit.sh
chmod +x ./update-talkkonnect-64bit.sh
/root/update-talkkonnect-64bit.sh

the talkkonnect.xml file was empty so I copied the latest version.

talkkonnect, 2023-04-09:
So you successfully updated?

f59465-Clem, 2023-04-10:
Yes successfully update .

f59465-Clem, 2023-04-10:
To return to the original subject of the Announcement played station on the RF side with the voiceactivity GPIO set to high, which sends the audio from a .wav file to the radio.

The goal is to send a looping audio file every x minutes on the Radio RF side for an announcement intended to inform users on radio waves only.

In the API I can't find a command to activate the voiceactivity GPIO, nor a command to broadcast an audio directly, apart from playback, but it needs two actions.
 
I thought of modifying the beacon function, keeping the same setting parameters and authorizing sending only on the RF side (GPIO voiceactivity).
Follow, did you have a trick to do this or is it more subject to a request for a new function ?

Thanks

talkkonnect, 2023-04-11:
To play a looping file every x minutes use beacon function. No need api.

f59465-Clem, 2023-04-11:
Hi Suvir,

So I did a test with "Beacon", unsurprisingly the beacon function works, on the other hand subtlety it sends the audio only on the mumble side (gpio txptt) not on the RF side (gpio voiceactivity).

Is there a possibility to modify this?
Or add an option in a future release to choose gpio txptt or voiceactivity or both?

Concerning "multimedia section" I do not understand how it works I did not find any help on this subject, how to make it work ?

f59465-Clem, 2023-04-17:
Hi Suvir,

I'll give you a feedback on the use of the beacon function locally.

 Overall it works, however there is a weird operation, if the beacon is triggered while a person is talking from Mumble, the beacon will be played at the same time ok, however at the end of the beacon it cuts the audio on the mumble side, hang very long until a new operator speaks. 

I'm not sure it's clear.

I think we should put a condition on the audio tag, if audio since mumble no tag played until next time.

Cordially
Clem