talkkonnect/talkkonnect #85: Feature Request - Power Off Button through GPIO

mudislandkid, 2023-04-14:
Forgive me if this already exists, but I couldn't find any documentation on it or through looking through the code.

I think it would be great to be able to 'safely' terminate talkkonnect and then shutdown the RPI through the use of a GPIO connected momentary button/switch. 

If I connect a momentary switch to GPIO 3 and GND I can through using a python script for example, send a shutdown now command to the RPI which will then put it in a Low power state. I can then toggle GPIO 3 again to power it back up. The problem I've had is being able to use this script in conjunction with also triggering a proper termination of the talkkonnect script so that the user will hear the audible TTS command of talkkonnect shutting down. 

I've tried to send a ctrl+c command to the screen session via the script. This works in terms of it starting the cleanup() function, but the TTS never fires and the process then appears to end abruptly with the screen session closing. So I figured that maybe it might be better adding in another GPIO definition that would allow a user to specify a length of time (say 5 seconds) that they want to have to hold down the button for, before it then triggers the shutdown process of talkkonnect, followed by an issuing of 'shutdown now'.

It needs to be GPIO pin 3 as that is the only pin that is hardware encoded to signal a 'startup' command when triggered.

I think this would be beneficial across multiple use cases and ensure that no one accidentally corrupts their SD card from too many hard shutdowns!

talkkonnect, 2023-04-15:
I can create a function for GPIO to map a bash command that can be defined in XML,  not only shutdown, you can map any bash command I think that would be more flexible. I think this will add to the configurability and flexibility of talkkonnect.
 
My question is how will you turn it back on how will you turn in back on?

mudislandkid, 2023-04-15:
That sounds interesting! I suppose as long as the bash script can send a ctrl+c command to the running screen session and it behave normally then that would work. Problem I was having was that sending a ctrl+c keystroke programmatically to a running screen session didn’t behave normally. Talkkonnect wouldn’t terminate gracefully. It started to then just stopped abruptly. Which is why I was wondering if the quit talkkonnect function could be triggered internally via gpio. 

But a configurable script trigger via gpio would be very powerful. 

The rpi when in low power mode/halt status monitors gpio 3. When shorted to ground it will trigger the pi to boot up again. This is in the firmware level.

talkkonnect, 2023-04-15:
I have made a solution for gpio button mapped to shutdown of linux. However it was different than I said above. To get shutdown to work just add it in the pins section of the io in talkkonnect.xml

This is the config that works below.

pin direction="input"  device="pushbutton"    name="shutdown" pinno="26" type="gpio" chipid="0" enabled="true"

If it works for you as it works here let me know and I will close this ticket.

mudislandkid, 2023-04-15:
Thank you. Will test it out this evening and let you know.

mudislandkid, 2023-04-15:
So I've been playing for a little while to try get it to work. I didn't do a full update using the script as that broke it for me last time. so I copied across the files you had made changes to and then did a rebuild.

When I run the build command however I get this error: -

```
# github.com/talkkonnect/talkkonnect
./client.go:361:5: b.beaconPlay undefined (type *Talkkonnect has no field or method beaconPlay)
```

Despite this error, I can still run talkkonnect, however it gives me the following error about the GPIO name: -

```
[  warn ] 2023/04/15 22:27:33 Config Error [Section GPIO] Enabled GPIO Name shutdown Pin Number 26 Invalid Name                                   
[  warn ] 2023/04/15 22:27:33 Non-Critical Errors Found In talkkonnect.xml config file please fix errors or talkkonnect may not behave as expected
```

Also, will this work if I map the gpio pin to GPIO pin 3? Or will it say that it conflicts with the OLED? It has to be GPIO 3 or else the push button won't restart the RPI once it has shutdown. Also can I adjust how long the pushbutton has to be held down for before it triggers the shutdown event? Ie 5 seconds or something?

Many Thanks!

talkkonnect, 2023-04-16:
The code works for sure. I have tested it. You will have to update using go get or update script. You are missing some lines in sanity checker etc from the errors you described.

i2c 1 uses gpio3 for oled so you may have to use another set of pins for i2c if you gave to use GPIO 3 for power up and shutdown.

For the 5 second thing no we do not support that long short keypress for different commands yet.

mudislandkid, 2023-04-16:
Ok that works now. However just a note, for some reason whenever I use the update script (32bit) on a build from source build, it never seems to copy back the xml file. It successfully creates a backup of my current talkkonnect.xml in the root directory, but then doesn't copy it back so have to do that manually. Not a big issue, but just making you aware.

So when the shutdown is set to pin 26 it works great. When I set it to pin3, I get a warning about an invalid GPIO pin, but it still kind of works. It will shutdown and send a sigkill but I only get a fraction of the TTS saying quit talkkonnect and then the pi shutsdown. 

Is there a way you can allow both the I2C and the shutdown pin to use GPIO3? This should work and not cause any issues with the OLED. That way I can just use one button to both shutdown and startup the dpi, rather than needing to separate ones.

Thank you again for all your work on this! Really is appreciated!

mudislandkid, 2023-04-16:
However....now the script that I wrote in python works as it should! So whatever you did to the cleanup() function seems to have made the difference!

This code allows me to set a 'hold' time for the button. so it has to be held down for 3 seconds before it will initiate the quit talkkonnect and then the system shutdown. This is with shorting GPIO pin 3 to ground. Works a treat.

Here is my python code for reference. Thanks again

```
import RPi.GPIO as GPIO
import time
import os
import signal

# Define the GPIO pin number for the button
BUTTON_PIN = 3

# Set up GPIO
GPIO.setmode(GPIO.BCM)
GPIO.setup(BUTTON_PIN, GPIO.IN, pull_up_down=GPIO.PUD_UP)

# Detect button press and hold for 3 seconds
button_pressed_time = None
while True:
    button_state = GPIO.input(BUTTON_PIN)
    if button_state == GPIO.LOW:
        if button_pressed_time is None:
            button_pressed_time = time.time()
        elif time.time() - button_pressed_time >= 3:
            break
    else:
        button_pressed_time = None
    time.sleep(0.1)

# Check if screen session is running
screen_session_exists = os.system("screen -ls | grep talkkonnect-radio") == 0

# Send Ctrl+C to screen session if it exists
if screen_session_exists:
    os.system("screen -S talkkonnect-radio -X stuff '\x03'")  # Send Ctrl+C character

# Pause for 10 seconds
time.sleep(10)

# Send shutdown command to Raspberry Pi
os.system("sudo shutdown now")
print("Would shutdown now")
```

talkkonnect, 2023-04-16:
Mudislandkid,
glad you got what you wanted. As for the talkkonnect.xml I prefer people to copy it back themselves I do not want to overwrite someones config so yes it is intentionally set for people to copy it manually.

Allow me to close this ticket.

closed 2023-04-16 by talkkonnect