talkkonnect/talkkonnect #86: Nvidia Jetson Nano GPIO

umekzg, 2023-04-15:
Has anyone tried with Jetson Nano?

I'm curious does it work out of the box or are there any problems with GPIO?

talkkonnect, 2023-04-15:
I don’t have that board and so far no reports from users reporting tests with Jetson Nano. Doing some reading online some articles say it’s pin for pin compatible where software is concerned, as for hardware the jetson will need the help of a transistor switch because on output it does not have high output current like the raspberry pi does so you cannot drive a LED directly.

All this information I have got by reading online so it’s all untested and unverified.

should you have such a board and have the chance to run talkkonnect on the jetson nano please do let all of us know the results.

umekzg, 2023-04-15:
Switch for PTT is all i need, AFAIK pull up with 1k to 3V3 is enough to do the job.
I'll try and post my findings.

talkkonnect, 2023-04-16:
OK do let us all know i will keep this ticket open awaiting your experimental results.

umekzg, 2023-04-16:
Nvidia official image doesn't have alsa tools so i had to install them and configure my USB card.
Same procedure as on rpi.

Installation of tk was painless, just used tkbuild-64bit.sh.

It's working without sweat. 

I'll try the GPIO's tomorrow.

talkkonnect, 2023-04-16:
Umekzg,
If you get it to work please consider a creating bash script similar to tkbuild-64bit.sh in the scripts directory so that other members of the community can benefit from your contributions.

thanks again for your interest and engagement with the community

talkkonnect, 2023-04-19:
Allow me to close this ticket if there are any updates you can send me an email.

closed 2023-04-19 by talkkonnect