talkkonnect/talkkonnect #87: Would this work on the banana pi m2 zero

likegithubmomo, 2023-04-16:


talkkonnect, 2023-04-17:
Does talkkonnect work on bananapi boards?

Answer:
It Should work but you cannot use the images you will need to install from scripts as detailed below.
I never have had the opportunity to use a Banana Pi board before, so cannot say for sure.

You can try by following the steps below -
 
1. First installing a Debian based OS on the Banana Pi (Command Line With No GUI)
2. Check if ALSA is installed or not if not install it (For Audio)
3. First See if the OS you installed in 64 or 32 Bit then using the Raw version of one of the scripts provided in the scripts folder use either tkbuild-32bit.sh or tkbuild-64bit.sh using wget to your root user home folder. 
4. chmod +x the script to make it executable
5. run the script for the installation of talkkonnect

Let us know your results, so others using the same board can benefit from your experience.

talkkonnect, 2023-04-19:
Allow me to close this ticket if there are any more questions you can open a new ticket

closed 2023-04-19 by talkkonnect