talkkonnect/talkkonnect #88: return of use Announcement played

f59465-Clem, 2023-04-19:
Hi Suvir,

I'll give you a feedback on the use of the beacon function locally.

Overall it works, however there is a weird operation, if the beacon is triggered while a person is talking from Mumble, the beacon will be played at the same time ok, however at the end of the beacon it cuts the audio on the mumble side, hang very long until a new operator speaks.

I'm not sure it's clear.

I think we should put a condition on the audio tag, if audio since mumble no tag played until next time.

Cordially
Clem

talkkonnect, 2023-04-21:
Clem,
the media player is now set for non-blocking. The beacon should play without any problem now. However the concept of backing off if there is audio on the channel is not implemented yet I am very busy back to work at the moment. This solution will have to do for now. Please test it and let me know. For the switch also not now later.

f59465-Clem, 2023-04-21:
Suvir,
ok, thanks for the correction, I will update and test.
I will come back to the post.
Ok no problem for the future switch.
Cordially

f59465-Clem, 2023-04-23:
Suvir,
The modification broke the code of the beacon function, the beacon does not trigger.
Cordially

talkkonnect, 2023-04-23:
Clem,
I have added back-off timer concept, so if talkkonnect receives a signal on stream it will reset the timer and the beacon will be re-scheduled to the next set interval. Released in version 2.23.05 Dated 23/04/2023. Update and test please.

f59465-Clem, 2023-04-24:
Suvir,
Thank you for the modification / addition of function, after updating with "update-talkkonnect-64bit.sh" the tag works, I will do tests for several days, I will report in a few days.

Cordially