talkkonnect/talkkonnect #89: Pi Zero 2W 64bit usb image "talkkonnect loaded" loop

coffeebird, 2023-05-21:
Using the latest talkkonnect-v2-rpi3-64bit-usbsound image with a pi zero 2W

Usb soundcard: https://www.aliexpress.com/item/4000598865391.html

I can hear "talkconnect loaded", but no sound of connecting to anything using either the default xml or the one I use in a pi3b to connect to my mumble server.

Instead, there's just a loop of "talkkonnect loaded" every 20 seconds or so.

talkkonnect, 2023-05-21:
looks like your raspberry pi is not connected to the wifi. This behaviour usually means that talkkonnect cannot connect to the server so it keeps trying again and again.

talkkonnect, 2023-05-22:
I will close this ticket as there  was a wifi connectivity problem here.

closed 2023-05-22 by talkkonnect