talkkonnect/talkkonnect #90: problem with Talkkonect service

MirkoUgoliniDev, 2023-05-27:
Hello all, 

after  while I got that error 

pi@raspberrypi:~ $ journalctl | grep  FAILURE
mag 27 14:28:33 raspberrypi systemd[1]: talkkonnect.service: Control process exi                ted, code=exited, status=1/FAILURE
mag 27 14:29:18 raspberrypi systemd[1]: talkkonnect.service: Control process exi                ted, code=exited, status=1/FAILURE
mag 27 14:33:46 raspberrypi systemd[1]: talkkonnect.service: Control process exi                ted, code=exited, status=1/FAILURE
mag 27 14:34:15 raspberrypi systemd[1]: talkkonnect.service: Control process exi                ted, code=exited, status=1/FAILURE

How can I investigate the problem causing these errors?

Sometime the RPI4 shut down

Thanks in advance

talkkonnect, 2023-05-28:
Look at the actual talkkonnect logs ssh in as root and do screen -r and see when the problem happens or alternatively log to file by the setting the logging to file in  xml config modifying the 3 XML tags below
 
xml tag logfilenameandpath = /var/log/talkkonnect.log
xml tag  logfilenameandpath/=  var/log/talkkonnect.log
xml tag  logging = screenandfilewithlineno

talkkonnect, 2023-06-05:
since no further response or questions I will close this ticket.

closed 2023-06-05 by talkkonnect