talkkonnect/talkkonnect #91: documentation for registering talkkonnect clients and using multiple shout-/voicetargets

sawatodayo, 2023-06-05:
while a basic setup with joining a channel and talking works, I can not figure out how to generate and use a certificate on raspberry pi talkkonnect clients and use it to program shout buttons. I'd like to request an addition to the documentation on how to:
- generate certificates for talkkonnect clients on raspberry pi and use them
- set different channels or users as shout-/voicetargets
- assign different gpio push-to-talk buttons to different voicetargets

in your current documentation it's already mentioned, that whisper/shout is working in talkkonnect, but never really referenced on how to set it up.

Thanks

talkkonnect, 2023-06-06:
I will update the documentation as per your request when I have some free time. To generate certificates there is a script called gencert.sh in scripts directory. You then need to add the absolute path to your certificate in the talkkonnect.xml file.
The xml tag is called certificate  and is in each account.

Once you generate the certificate and configure the xml file you will need to set the register tag to true to register to your server.

After the registration process the voice targets will work.

For voice targets you will have to set the id tag and the username you want to whisper to in the xml config. You then reference your id tag in the pins section of your gpio.

sawatodayo, 2023-06-06:
thanks. was kind of confused with the process. thank you for the fast answer

sawatodayo, 2023-06-06:
I read through all available code and didn't find any function to map voicetargets/shout to GPIO. There are commands for MQTT and USB-keyboard keys, but not even in gpio.go I can find a command for sendVoiceTargets. Can you please explain in more detail how to do it or, if you forgot to implement it, add it to the code? I guess it would be a important feature to be able to speak to multiple channels for many radio/intercom projects

Thank you

talkkonnect, 2023-06-07:
Hi Sawatodayo,

Thank you for your follow up. I now can remember that the voice targets was not implemented with GPIO, 
they work with HTTPAPI, MQTT and USB keyboard.

The reasoning behind this was that there are 31 voice targets to be set and if even for the use case
where there are lets say 5 voice targets that would require 5 GPIO to set the voice target and 1 GPIO
to clear out the voice targets. I came to the conclusion that we do not have enough buttons and no
one would do that.

However with a USB keypad (Wired or Wireless) there is a cheap solution that will work for headless 
machines in which you have enough keys to play and set as you wish.

The USB keyboard directly talks to talkkonnect throught device driver so there is no need to log in
the USB keboard will work like GPIO. 

The advantages of USB keyboard are
1. Enough Buttons
2. Can set any button for any function
3. Very Affordable and Cheap
4. No need to build any hardware
5. Can use remaining GPIO for other more necessary functions

I am grateful that you took the time to look though the code, not many people do this and I have tried
to make the code readable and easily modified. Should you like to implement this feature in GPIO please
do contribute back the code for others with the same requirements.

Best ,

Suvir

sawatodayo, 2023-06-07:
That explains it. Most of commercial intercom solutions have 4 channel buttons. That's what I'm trying to replicate. I'm trying to code it myself, but I seem to not be able to add gpio functions. I added the variables, initialization (pullup etc), rpi.close function (the long row), and the function of the button (copied the base code from memorychannel function) in gpio.go. Despite all that it does not even send me a "button pressed" debug information. config xml is also appended with input-pushbutton-[name of function]-pinno-gpio etc. Am I missing a function I should add?

talkkonnect, 2023-06-07:
I can add 4 voice target to GPIO if you want. Shouldn’t be too difficult. 4 buttons would mean 3 voice targets and 1 button to clear the voice targets.
Channel buttons are already implemented in GPIO. But I could do the work and add 4 voice target buttons. Let me know.

sawatodayo, 2023-06-07:
I would really be grateful for that. It would probably make my (and other people's) life easier. I don't know why, but I just can't get the GPIO to run.

Thank you very much

talkkonnect, 2023-06-07:
So other GPIO like tx button and rx led etc are working for you correct? What version of raspberry pi are you using?

talkkonnect, 2023-06-11:
Sawatodayo,
version 2.25.01 released 11 june 2023 supports 5 gpio voice targets buttons. Please read the file version.go for the notes on how to configure in xml.

talkkonnect, 2023-06-13:
Since the new feature is added and locally tested and no further response from community . I will close this thicket.

closed 2023-06-13 by talkkonnect