talkkonnect/talkkonnect #92: Bugs Talkkonnect

f59465-Clem, 2023-06-21:
Hello everyone,

Version 2.23.05
I notice problems with:

"**txtimeoutsecs**" does not work despite txtimeout = true, during a blockage does not cut.

"**txptt**" the activity of the GPIO pin is no longer reported in the logs (Bug or option).

"**beacontimersecs**" the beacon does not work when the seconds are greater than 900sec (15min), also strange operation to have 15min it is necessary to put 600sec or 10min.

greeting
Clem

talkkonnect, 2023-06-22:
Clem,
thank you for the bug report.
1. txttimeoutsecs is indeed not working 
2. txptt is now only visible when debug level is debug and is not showing in info level of debugging I plan to make the info mode more quiet and will be removing more tags soon
3. for beacontimersecs I have not set any limits on time time so its strange that there is a 15 min max interval this was not intentional and I cannot reproduce it.

when I get time will try to fix out the bug item # 1

suvir

f59465-Clem, 2023-06-26:
Hello Suvir,

in the immediate future it is bug of "txttimeoutsecs" which poses the most problem, without cut (TOT), the use of TK in RF / Voip gateway poses big problems, because if problem the other RF / Voip relays interconnect remains blocked in transmission. I prefer to put the relay offline when I'm not nearby to avoid the risk of blocking the network (4 relay rf/voip) I'm the only one using TK.

For the beacon problem, it's not systematic, it happens after the 2-3rd time, and after that all the TK LEDs go off, a reboot is needed until the next bug.

I will try on my side to reproduce the bug of "beacon".
I think fixing txttimeoutsecs is a priority.

Sincerely

talkkonnect, 2023-06-27:
Dear Clem,
Maybe it is unclear but for your use case. To implement a timer on the output of talkkonnect to keying the RF radio you
need to set    txlockenabled to true and txlocktimeoutsecs to 30 for example if you want the radio to key maximum 30 seconds.

Can you check and clarify this? maybe the tagnames are confusing we do not use txtimeoutsecs

Try and let me know the feedback.

suvir

talkkonnect, 2023-06-27:
Dear Clem,
latest version released removed txtimeout as it is confusing please use txlocktimeout instead.

f59465-Clem, 2023-06-27:
Hi Suvir,
I tried txlockenabled to true and txlocktimeoutsecs which I found in "panicfunction" it doesn't work.
No TOT on the TX.

Clem

talkkonnect, 2023-06-29:
Clem,
Let me clarify you are looking for a timeout on the voiceactivity LED on RX. You are using this signal to key up the RF transmitter and you want to turn off this voiceacvitivy LED if the time exceeds a certain threshold so that talkkonnect will stop keying the transmitter even if someone is still talking is that what you want? However if that person stops talking on the mumble side and starts another new long transmission it will wait for the next lockout time correct?

This was planned in the XML file but never implemented. I have yet to think of an elegant solution how this should be handled.

As I have not yet used talkkonnect as a repeater myself all that was created was from theory Please advise.

suvir

f59465-Clem, 2023-06-29:
Hi Suvir,

  Yes, like a Baofeng TOT, etc.
  Control of the transmission time, if it exceeds it hangs up, or if the GPIO remains blocked it also hangs up.

Clem

talkkonnect, 2023-06-29:
The elegant way now is to disable the voiceactivity LED, until it is re-enabled manually by repeater controller?

f59465-Clem, 2023-06-29:
Specifically, I request this function for TXptt (TX of TK<-RX RF).

But yes both ways for Voiceactivity and txptt would be ideal with an adjustable timer for each, (RX TK -> TX RF & TX TK<- RX RF).

Skipping the timer hangs up until the person presses PTT again.

Clem

talkkonnect, 2023-08-08:
OK I have put this on my todo list will do it when I have some time. It is not a small effort to do this as the ptt logic is quote complicated. I will close this bug report for now.

closed 2023-08-08 by talkkonnect