talkkonnect/talkkonnect #93: Recording combo 24/7

s3m1s0n1c, 2023-08-09:
Hey Guys,

Love Talkkonnect so thanks for all your work..

Ive been playing with avrecord.go to try and make it so it records 24/7 and when theres like x amount of silence (being set int xml file) then it will create a new file..

**Basically wanting recording for each transmission in a new mp3 when silence is like 5 seconds**

This is my code so far:

Thanks
Sonic

`func audiorecordcombo() {
    checkfile := isCommandAvailable("/usr/bin/sox")
    if !checkfile {
        log.Println("error: sox is Missing. Is the Package Installed?")
        return
    }

    audrecfile := time.Now().Format("20060102150405") + "." + Config.Global.Hardware.AudioRecordFunction.RecordFileFormat
    log.Println("info: sox is Recording Traffic to", Config.Global.Hardware.AudioRecordFunction.RecordSavePath+"/"+audrecfile)
    log.Println("info: Audio Recording Mode:", Config.Global.Hardware.AudioRecordFunction.RecordMode)

    if Config.Global.Hardware.AudioRecordFunction.RecordTimeout != 0 {
        args := []string{"-q","-m", "-t", Config.Global.Hardware.AudioRecordFunction.RecordSystem, Config.Global.Hardware.AudioRecordFunction.RecordFromOutput, "-t", Config.Global.Hardware.AudioRecordFunction.RecordSystem, Config.>

        log.Println("debug: sox Arguments: " + fmt.Sprint(strings.Trim(fmt.Sprint(args), "[]")))
        log.Println("info: Audio Combo Recording will Timeout After:", Config.Global.Hardware.AudioRecordFunction.RecordTimeout, "seconds")

        cmd := exec.Command("/usr/bin/sox", args...)
        cmd.Dir = Config.Global.Hardware.AudioRecordFunction.RecordSavePath
        err := cmd.Start()
        if err != nil {
        log.Println("error: Failed to start sox command:", err)
        log.Println("stderr:", stderr.String()) // Print the standard error output
        return
        }
        check(err)
        done := make(chan struct{})

        time.Sleep(time.Duration(Config.Global.Hardware.AudioRecordFunction.RecordTimeout) * time.Second)

        go func() {
            err := cmd.Wait()
            status := cmd.ProcessState.Sys().(syscall.WaitStatus)
            exitStatus := status.ExitStatus()
            signaled := status.Signaled()
            signal := status.Signal()
            log.Println("error: sox Error:", err)
            if signaled {
                log.Println("debug: sox Signal:", signal)
            } else {
                log.Println("debug: sox Status:", exitStatus)
            }
            close(done)
            log.Println("info: sox Stopped Recording Traffic to", Config.Global.Hardware.AudioRecordFunction.RecordSavePath)
        }()
        cmd.Process.Kill()
        <-done

    } else {
        audrecfile := time.Now().Format("20060102150405") + "." + Config.Global.Hardware.AudioRecordFunction.RecordFileFormat
        args := []string{"-q","-m", "-t", Config.Global.Hardware.AudioRecordFunction.RecordSystem, Config.Global.Hardware.AudioRecordFunction.RecordFromOutput, "-t", Config.Global.Hardware.AudioRecordFunction.RecordSystem, Config.>
        log.Println("debug: sox Arguments: " + fmt.Sprint(strings.Trim(fmt.Sprint(args), "[]")))
        cmd := exec.Command("/usr/bin/sox", args...)
        cmd.Dir = Config.Global.Hardware.AudioRecordFunction.RecordSavePath
        err := cmd.Start()
        if err != nil {
        log.Println("error: Failed to start sox command:", err)
        log.Println("stderr:", stderr.String()) // Print the standard error output
        return
        }
check(err)
        time.Sleep(2 * time.Second)

        if err == nil {
            log.Println("info: sox is Recording Mixed Audio to", Config.Global.Hardware.AudioRecordFunction.RecordSavePath)
            log.Println("warn: sox will Go On Recording, Until it Runs out of Space or is Interrupted")

            starttime := time.Now()

            ticker := time.NewTicker(30 * time.Second)

            go func() {
                for range ticker.C {
                    checked := time.Since(starttime)
                    checkedshort := fmt.Sprintf("%v", before(fmt.Sprint(checked), "."))
                    elapsed := fmtDuration(checked)
                    log.Println("info: sox is Still Running After", checkedshort+"s", "|", elapsed)
                    
                    if checked >= 5*time.Second {
                        log.Println("info: Creating a new audio file due to 5 seconds of silence...")
                        cmd.Process.Signal(syscall.SIGINT)
                      
                        audrecfile = time.Now().Format("20060102150405") + "." + Config.Global.Hardware.AudioRecordFunction.RecordFileFormat
                        args[15] = audrecfile // Update the filename in the arguments
                        cmd = exec.Command("/usr/bin/sox", args...)
                        cmd.Dir = Config.Global.Hardware.AudioRecordFunction.RecordSavePath
                        err := cmd.Start()
                        check(err)
                        starttime = time.Now()
                    }
                }
            }()
        } else {
            log.Println("error: Something Went Wrong... sox Traffic Recording was Launched but Encountered Some Problems")
            log.Println("warn: Check ALSA Sound Settings and sox Arguments")
        }
    }
}

`

talkkonnect, 2023-08-09:
Hi Sonic,
the recording module was developed by Zoran. So far all I did was clean it up. I have never tested the recording to be honest. However I will take the time to test it out and see and feedback time permitting.

always appreciate good feedback,

suvir

s3m1s0n1c, 2023-08-10:
Thanks @talkkonnect :)

s3m1s0n1c, 2023-08-18:
@zrdimetc Thanks, but I have done all this..
The sox command runs fine manually..

The problem is that the avrecord.py doesn't really work for what is was written for which is problem 1 and problem 2 is I would like it to record each transmission 24/7 

Hence why I have put in a issue request..

zrdimetc, 2023-08-18:
SoX and Alsa sound loopback recording function is just an example how to call an external utility from talkkonnect to do a basic recording to a file. It continuously records audio to some maximum file size that you find manageable, then starts a new file and archives old recording. This served a purpose of demonstrating how to use SoX with talkkonnect. Changing the logic to detect individual transmissions based on silence and record them to individual files can be tricky, but with more time, effort and contributions to the recording code improving, this should be possible. 
If the basic recording settings from` talkkonnect.xml` are not working for you,  `xmlparser.go` version 2 should be inspected together with `avrecord.go`.  Recording function was last tested with version 1 of talkkonnect.

zrdimetc, 2023-08-18:
As long as you made your sound subsystem recording working with Alsa loopback or pulseaudio, you could try to manage your audio recording with an external script outside of talkkonnect? You could start audio recording with SoX detecting silence and then keep checking the recorded file size by an external process like `du`. When the file stops growing it means the silence is detected? At that moment an external script can kill old SoX and start a new SoX instance? Doing this in talkkonnect would require many changes. talkkonnect would have to open and sanitize SoX instances, over and over again. This would be work intensive and there are many things that could possibly go wrong and affect talkkonnect stability. If anyone else could contribute to improving the recording function, that would be great.

zrdimetc, 2023-08-19:
If SoX returned non-zero exit status 2 or you see an error in the talkkonnect console
"`avrecord.go:201: Something Went Wrong... sox Traffic Recording was Launched but Encountered Some Problems`" ,
it means some of the SoX arguments in talkkonnect config file were wrong or missing.

For making the recording function work as it is, you need to give the sound system argument to sox - `alsa` or `pulseaudio`. If you don't want to play with Alsa loopbacks and Alsa plugins, I suggest use pulseaudio and a monitor device. It will be more easy to make the recording work. Find out your monitor device name with pactl or pacmd.

talkkonnect recording config could look like this for continuous traffic recording.
```

<audiorecordfunction enabled="true">
<recordonstart>false</recordonstart>
**<recordsystem>pulseaudio</recordsystem>**
<recordmode>traffic</recordmode>
<recordtimeout></recordtimeout>
**<recordfromoutput>alsa_output.usb-C-Media_Electronics_Inc._USB_PnP_Sound_Device-00.analog-stereo.monitor</recordfromoutput>**
<recordfrominput></recordfrominput>
<recordmictimeout></recordmictimeout>
<recordsoft>sox</recordsoft>
<recordsavepath>/avrec/audio</recordsavepath>
<recordarchivepath>/avrec/archive/audio</recordarchivepath>
<recordprofile></recordprofile>
<recordfileformat>mp3</recordfileformat>
<recordchunksize>600</recordchunksize>
</audiorecordfunction>

```
Recording doesn't start automatically in this example, it is called with "Ctrl-I".

```
[  info ] 2023/08/19 01:55:47 commandkeys.go:598: Traffic Recording Requested
[  info ] 2023/08/19 01:55:48 avrecord.go:134: sox is Recording Traffic to /avrec/audio/20230819015548.mp3
[  info ] 2023/08/19 01:55:48 avrecord.go:135: Audio Recording Mode: traffic
[  info ] 2023/08/19 01:55:50 avrecord.go:185: sox is Recording Traffic to /avrec/audio
[  info ] 2023/08/19 01:55:50 avrecord.go:186: sox will Go On Recording, Until it Runs out of Space or is Interrupted
```

You should at this point be able to see the recorded audoo file being written in "_recordarchivepath_".
Old recordings are packed and archived.

talkkonnect, 2023-08-20:
Hey Zoran, Hey Sonic,
I have released talkkonnect version 2.29.01 just now supporting sanity checking for alsa and pulseaudio for sound system and also fixed the switches to work with sox as solution recommended by Zoran.

Please test it and feedback.

Suvir

zrdimetc, 2023-08-20:
Let me leave few more comments about how the recording logic was first designed to work, in addition to the [blog article](https://www.talkkonnect.com/2020/04/18/398/) and comments in avrecord.go. 

Originally, the recording was designed to work only with Alsa. Since then, pulseaudio is supported, hence the need for defining the sound system in talkkonnect config, to tell SoX binary, what sound system to talk to (the `recordsystem` tag). File with recording functions was called “`av`” (_audio and video_) recording. We wanted to record both audio and video, and in addition to `sox`, support `ffmpeg`, `cvlc` or other utilities capable of recording. Talkkonnect has a “_panic button_” function which actuates the PTT button and broadcasts ambient audio to a Mumble channel. Taking video or an image snapshot in emergency situation was a logical next step in developing that kind a of a function. `avrecord.go` was meant to serve as an repository of all functions for interfacing between talkkonnect / Mumble and external recording utilities.
 
- When it comes to recording, user have different needs, resources available to them, preferred logic, understanding of Linux sound and types of audio they want to record. Some people record voice, some music. Some audio streams are loud and dynamic, other... unpredictable. For some people it is important to conserve the recording media space. SoX has features like detecting silence, trimming the recordings and applying all kinds of effects and manipulations. SoX can also mix recording from different sources.

- Currently, there are **3 recording modes** in talkkonnect. 1. Record sound output - Mumble channel traffic, (`traffic` mode) 2. Record ones’ own talking and ambient sounds (`ambient` mode). Or 3. Combine the 2 recordings (mic and speaker from 1. and 2.) in a single track (`combo` mode). This depends on knowing your sound system devices, their names, understanding SoX arguments and deciding what is it that you want captured?

- When it comes to "where" to record, it is recommended not to record on a system SD card to minimize it’s wear and tear. Instead, use USB stick or drive and mount it. Recording to a shared drive, FTP server, cloud storage - are all the possibilities that come to mind.

- On talkkonnect startup or calling the recording function with keyboard shortcut (`Ctrl-I, Ctrl-J, Ctrl-K`) or http api call, old recordings will be zipped and moved to a predefined archive folder. If users do not need the archiving, they can change the recording and comment archiving function out.

- Some people want to record continuously and some only want to start and stop recording from time to time. Either way, when the recording is initiated, we don’t want the recorded files to grow too large and become hard to handle or consume the disk space entirely. Currently, talkkonnect is not checking if the destination media has space for recording or not. No space, or if config is bad or arguments for SoX missing, recording won't work.

Therefore, the options to “`timeout`” the recording once it starts, or trim and breakdown files to manageable “_chunks_”. 
At the same time, if we allow recording to grow indefinitely, we though that "silence", "trim" and "newfile" with "restart" were useful SoX options, to keep the recorded files under control.

If the recording timeout tag in talkkonnect config is set to “**0**” this will make the recording continue indefinitely, for as long as there is free space. If “trim” is used together with “newfile”, “restart” arguments and “chunk” file size set in the config, then the file sizes will be limited. You will have the resulting files files trimmed to a predegfined length, e.g. 100's of files each containing 10, 30, 60 minutes of audio (defined under "chunks" as seconds, 600, 1800, 3600, etc)

- When it comes to the recording logic, the way recording works is more similar to how Mumble client recording works – can be initiated as needed, and go on continuously. Files will have a time stamp. There is no function to stop recording with a keyboard shortcut at this time. For stopping the recording kill sox or restart talkkonnect with desirable changed config. Stop the recording keyboard shortcut may be added later.

- Can talkkonnect and SoX recording work like Zello? To record each individual transmission to a separate file with time stamp and name of who was talking? That should be possible, since talkkonnect has a go channel providing info on who is talking and info about stream starting and stopping can be used to call SoX and name the recorded file accordingly. SoX will need more strict control from talkkonnect and better error handling.

- In some new revision of the recording functionality, Suvir will probably make a decision to simplify the recording logic. SoX arguments for silence, trimming, newfile, restart, will no longer be hardcoded in avrecord.go and basic recording configuration options will be simplified. Instead, the users will have to enter their entire preferred SoX arguments line in one go - to talkkonnect config file. At that time, Suvir can consider using the stream info for controlling the SoX recording. 
In the mean time, users requiring recording logic for SoX or some other utility like FFmpeg, in a different way from how talkkonnect recording `avrecord.go` template works at this time are advised to create and apply their own scripts to control external recording programs or to try to customize the existing `avrecord.go` with to their specific recording requirements. 

**talkkonnect** is a real “transformer-toy", “lego bricks” set, a “swiss pocket knife” for PTT IP communications. The framework that Suvir designed, makes it easy for everyone to customize talkkonnect functions to their own liking or add completely new functions and integrations at ease to unlock the potential of PTT communications for each case.

zrdimetc, 2023-08-21:
This is recorded Mumble traffic (mode `traffic`). Each recording is exactly 10 minutes long - configurable with `recordchunksize` tag, then SoX records to a new file. Recording goes on continuously for as long as talkkonnect is running or there is free space. Recording can be initiated automatically on talkkonnect launch with `recordonstart `tag or started with `Ctrl-I` shortcut.
[image]
On a talkkonnect new start, recorded files will be archived and if recording is enabled, the new files will be recorded to the same path `/avrec/audio`. Files from previous talkkonnect run will be moved to `/avrec/archive/audio`... or wherever it is specified in talkkonnect config file. Enabling or disabling archiving will be managed in config in future. 
[image]
This is how recording of traffic works.

zrdimetc, 2023-08-22:
Archiving function for the old recorded files will be commented out for now in `avrecord.go`. If archiving and recorded files management are required, it can be done by an external script.

talkkonnect, 2023-08-26:
Sonic,

I have updated the avrecord.go to properly support recording before the next iteration.
Please test and feedback.
Let me close this ticket.

Thank you Zoran for the contributions to the project.

suvir

closed 2023-08-26 by talkkonnect

s3m1s0n1c, 2023-08-28:
Thanks everyone for your comments.

Thanks @talkkonnect for fixing the recording..

It now works well and fit for purpose :)

Sonic

zrdimetc, 2023-08-29:
The recording section of `talkkonnect.xml` for recording just the incoming Mumble traffic. Output device that SoX is listening to is a pulseadio monitor. Files are saved as 1 hour audio in mp3 format.
```
<audiorecordfunction enabled="true">
<recordonstart>true</recordonstart>
<recordsystem>pulseaudio</recordsystem>
<recordmode>traffic</recordmode>
<recordtimeout>0</recordtimeout>
<recordfromoutput>alsa_output.usb-C-Media_Electronics_Inc._USB_PnP_Sound_Device-00.analog-stereo.monitor</recordfromoutput>
<recordfrominput></recordfrominput>
<recordmictimeout></recordmictimeout>
<recordsoft>sox</recordsoft>
<recordsavepath>/avrec/audio</recordsavepath>
<recordarchivepath></recordarchivepath>
<recordprofile></recordprofile>
<recordfileformat>mp3</recordfileformat>
<recordchunksize>3600</recordchunksize>
</audiorecordfunction>
```
[image]

Recommend using USB stick with talkkonnect for recording. Format it with NTFS system `mkfs -t ntfs /dev/sdX` and mount it at boot with `/etc/rc.local` or `cron`.