talkkonnect/talkkonnect #94: multiple server

f59465-Clem, 2023-08-15:
Hi to all,

How to add a second server?
Can you do audio detection from one server to another?

Thank
Clem

talkkonnect, 2023-08-15:
Just add another XML block for the server and there are next server and previous server commands where you can change server.

talkkonnect, 2023-08-16:
This is an example of 3 servers

[image]

f59465-Clem, 2023-08-17:
Ok Suvir, 
I did like this before seeing the message, I have the CTRL+N "Next Server" command which does not work...

<accounts>
    <account name="community" default="true">
      <serverandport>server.com:10210</serverandport>
      <username>Relais_446</username>
      <password>****</password>
      <insecure>true</insecure>
      <register>false</register>
      <certificate></certificate>
      <channel>HAM-CB</channel>
      <ident>Name Surname</ident>
          <listentochannels>
                <channel>talKKonnect</channel>
          </listentochannels>
      <tokens enabled="true">
        <token>111111</token>
        <token>222222</token>
      </tokens>
      <voicetargets>
        <id value="0">
            <user>suvir-demo</user>
        </id>
        <id value="1">
          <users>
            <user>suvir-ubunbtu</user>
          </users>
        </id>
        <id value="2">
          <users>
            <user>suvir-demo</user>
          </users>
        </id>
        <id value="3">
          <channels>
            <channel>
              <name>TEST1</name>
              <recursive>true</recursive>
              <links>true</links>
              <group>"all"</group>
            </channel>
          </channels>
        </id>
      </voicetargets>
    </account>
	<account name="Server2" default="true">
      <serverandport>server_perso.net:64993</serverandport>
      <username>Relais_Perso_Test</username>
      <password>****</password>
      <insecure>true</insecure>
      <register>false</register>
      <certificate></certificate>
      <channel>HAM-CB</channel>
      <ident>Name Surname</ident>
          <listentochannels>
                <channel>talKKonnect</channel>
          </listentochannels>
      <tokens enabled="true">
        <token>111111</token>
        <token>222222</token>
      </tokens>
      <voicetargets>
        <id value="0">
            <user>suvir-demo</user>
        </id>
        <id value="1">
          <users>
            <user>suvir-ubunbtu</user>
          </users>
        </id>
        <id value="2">
          <users>
            <user>suvir-demo</user>
          </users>
        </id>
        <id value="3">
          <channels>
            <channel>
              <name>TEST1</name>
              <recursive>true</recursive>
              <links>true</links>
              <group>"all"</group>
            </channel>
          </channels>
        </id>
      </voicetargets>
    </account>
  </accounts>

talkkonnect, 2023-08-27:
Clem,
I have found issue with next server and have fixed it now it is mapped to key ctrl-Q for next server make sure you have the tag nextserver index in your talkkonnect.xml file.

updated version fixed bug and released in version  2.31.01 Rel 27 Aug  2023

thanks for your bug report,
suvir kumar

closed 2023-08-27 by talkkonnect

talkkonnect, 2023-08-27:
[image]

f59465-Clem, 2024-01-16:
Reopening.