talkkonnect/talkkonnect #95: RF repeater CTCSS

f59465-Clem, 2023-08-25:
Hello everyone,

I saw that Talkkonnect handles CTCSS for DRA/SA818 modules.
And you it possible to add this function for use RF repeater, because if repeater on remote site of 200km to complicate to change the ctcss of the radio station quickly, for problems of management of abuse, to facilitate management, others.

Support for standard CTCSS with a hotkey to quickly ON/OFF the ctcss.

Thanks

talkkonnect, 2023-08-26:
So you want to control an external ctcss decoder correct? How is this external ctcss decoder programmed or how does the user change the tone? Is it currently via dip switch?

f59465-Clem, 2023-08-27:
Hi Suvir,

I think I was too confused, I'll try to be more specific.

It would be great to add a CTCSS software decoder to Talkkonnect when using Talkkonnect as an RF gateway.

  To have a remote control of the ctcss via Main with a keyboard shortcut to activate it or not, thus avoiding using the CTCSS decoder function on the RF station side and thus avoiding all the problems mentioned in the previous message.

(Example of Hamradio software "SVxlink" which includes a software ctcss decoder.)
[https://www..org/doc/man/man5/svxlink.conf.5.html](url)

Thanks Clem

talkkonnect, 2023-08-27:
Hi Clem,
We are currently using alsa with talkkonnect, I think if we want to listen to the stream from another thread we need to use pulse audio, also I do not have enough programming knowledge to add this kind of frequency detection into talkkonnect by listening to the sound stream on rx. So as a quick win maybe you can use some external detector and provide some way that talkkonnect can send this external decoder the way to choose the ctss similar to sa818 which I think we can do.\
Let me know. However if there are any programmers out there who can help with this we will be most grateful.

suvir

f59465-Clem, 2023-08-27:
Ok Suvir,

Even if I managed to create a beacon with DTMF command with Arduino, I'm just an amateur in programming, here is some track for CTCSS tone detection it uses Goertzel's algorithm can also be used for DTMF command detection.
[https://www.embedded.com/detecting-ctcss-tones-with-goertzels-algorithm/](url)

After research on the hamradio voip gateway program, he most often uses ALSA another track would be multithreading.

Thank Clem

talkkonnect, 2023-08-28:
Let me know after your research for using external board if you can and the output with the rx signal from talkkonnect I guess you can key up the transmitter. You need a ctcss decoder you can control remotely.

talkkonnect, 2023-08-31:
let me know if there is any update for this request. i will close this ticket.

closed 2023-08-31 by talkkonnect