talkkonnect/talkkonnect #96: Voice activated tx possible?

seventer, 2023-08-25:
Hi All and sorry if already asked,

Is is possible to have a voice activated transmit?  The use-case is like babyphone, is sound is heard transmitting should occur.
A lot of clients have this feature but since running on a Pi, talkkonnect seems (by far) the best mumble client.

--Gerard

f59465-Clem, 2023-08-26:
Hi Gerard,
To date Talkkonnect does not support VOX, it seems to me that there is a solution, VOX in the search.
Sincerely

seventer, 2023-08-26:
Sorry to hear. I assumed since the ReSpeaker 2-Mics Pi HAT has this functionality already built-in it was accessible from talkkonnect.  My bad not investigating better before spending lots of money.
Thanks for your reply though.

talkkonnect, 2023-08-27:
There are external noise gate chips that can do COS. I will post a video when available from Boards available by mtech.

closed 2023-08-27 by talkkonnect