talkkonnect/talkkonnect #97: Error in install script tkbuild-32bit.sh

ahk97, 2023-09-05:
Script ./tkbuild-32bit.sh seems to run fine. However, it throws one error:

```
mv: cannot move '/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect' to a subdirectory of itself, '/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/talkkonnect'
=> Finished building TalKKonnect

```

talkkonnect, 2023-09-06:
Please use raspi-talkkonnect to build instead it is in the scripts directory. The build32 and build64 are depreciated. However the error you show above should not affect the building process. Can you try with rasp-talkkonnect and see its a menu driven build which is much simpler to use. Do let me know your results.

talkkonnect, 2023-09-07:
Thank you for the report I have fixed the tkbuild-32bit.sh and also the tkbuild-64bit.sh scripts based on your feedback.

closed 2023-09-07 by talkkonnect

ahk97, 2023-09-07:
Thank you!
I will also test the raspi-talkconnect later today

talkkonnect, 2023-09-09:
> raspi-talkconnect interactive script fails at menu item 3:
>
> getting 32 version of golang for cpu typearmv7l
> --2023-09-08 13:07:53-- https://go.dev/dl/go1.21.1
> Resolving go.dev (go.dev)... 216.239.32.21, 216.239.34.21, 216.239.38.21,
> ...
> Connecting to go.dev (go.dev)|216.239.32.21|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 264 [text/html]
> Saving to: â€˜go1.21.1.2â€™
>
> go1.21.1.2
> 100%[=========================================================================>]
> 264 --.-KB/s in 0s
>
> 2023-09-08 13:07:53 (1.37 MB/s) - â€˜go1.21.1.2â€™ saved [264/264]
>
> --2023-09-08 13:07:53-- http://time/
> Resolving time (time)... failed: Name or service not known.
> wget: unable to resolve host address â€˜timeâ€™
> --2023-09-08 13:07:53-- ftp://2023-08-31t22/36:09Z.linux-arm6l.tar.gz
> => â€˜36:09Z.linux-arm6l.tar.gzâ€™
> Resolving 2023-08-31t22 (2023-08-31t22)... failed: Name or service not
> known.
> wget: unable to resolve host address â€˜2023-08-31t22â€™
> FINISHED --2023-09-08 13:07:54--
> Total wall clock time: 0.4s
> Downloaded: 1 files, 264 in 0s (1.37 MB/s)
>
> gzip: stdin: not in gzip format
> tar: Child returned status 1
> tar: Error is not recoverable: exiting now
> ln: failed to create symbolic link '/usr/local/bin/go': File exists
> ln: failed to create symbolic link '/usr/local/bin/gofmt': File exists

If the symbolic links exist then it just means that this script has been
run before or that golang is already installed. There should be no errors
on a fresh os install when running this step item menu 3.

ahk97, 2023-09-11:
OK. Thanks.
I used the updated tkbuild-32bit.sh script and it was working fine.