talkkonnect/talkkonnect #98: Support for CM119A GPIO for Ham Radio Repeater Connection

litmus-paper, 2023-09-07:
Thank you so much for developing and supporting the project.

I am currently running a ham radio repeater network, connected using Allstarlink and Hamvoip. I am interested in transitioning this setup to use Talkkonnect.
My repeater's Push-to-Talk (PTT) and signal detector functionalities are connected via CM119A GPIO. Is there a way to configure Talkkonnect to work with CM119A GPIO for these functionalities?
I've gone through the documentation but couldn't find any specific guidance on this particular hardware. Any help or direction would be greatly appreciated.

talkkonnect, 2023-09-07:
CM108 can be used and detected as keyboard keypress. I believe its the media key or something like that. In talkkonnect you can map keyboard keypresses to ptt talkkonnect on the mumble side. For the Reverse if someone is talking on mumble there is a GPIO out also. So I am not sure about your question. Is the CM119A on the Allstarlink side? If the CM119A has gpio it can connect to talkkonnect via OPTO isolator for rx and tx also directly to raspberry PI running talkkonnect.

Maybe a diagram would help me understand what you are trying to do. I am sorry I am very ignorant about AllStarLink and HAMVOIP I dont know how they work or what they do.

Please explain and  I will try to help.

suvir

talkkonnect, 2023-09-07:
you can also talk to us on the HAM-CB channel which I standby on the community server.
address community.talkkonnect.com port 64738 no password channel HAM-CB.

litmus-paper, 2023-09-07:
I am providing a block diagram to explain the situation better:
[image: image.png]
A sound card based on CM119A is connected to the repeater. CM119A is the
sound card used to connect to the repeater. The repeater signal detection
is connected to the volume-up input on the audio card. I can use the
keyboard mapping to get this working. Thank you for the tip. The repeater
PTT input is connected to the CM119A GPIO output. How can I control it with
Talkkonect?
Do you happen to have any tips that I can share with a Go programmer to
help me do this?

This is a common setup since several ham-related sound cards adopted this
connection schema. For example, the URIxB is working this way. Here is its
schematics:
[image: image.png]

Thanks
Nir Eden
4X5NI

talkkonnect, 2023-09-08:
Nir,
I can’t see the image on GitHub can you send the images to my email instead suvir@talkkonnect.com

talkkonnect, 2023-09-09:
I think based on the block diagram you sent me. You can just do what everyone else who uses talkkonnect  as a repeater does. 

Talkkonnect provides GPIO pins on the raspberry pi for rx and tx already. So use the usb sound card for audio only. 

Connect the GPIO pins of raspberry pi preferably through optocoupler to the repeater to control its tx and in turn get its rx signal.

On the talkkonnect side there is a pin called voiceactivity which and be used to connect to the tx of the repeater. 

From the receive signal of the repeater you can connect to the pin marked as transmit on the raspberry pi. 

There is no need to use the GPIO pins provided by the sound card as you are using a raspberry with talkkonnect which already provides ready to use GPIO signals for interfacing to other devices.

These pins can be defined in talkkonnect.xml configuration file no changes are needed from the standard configuration usb sample provided in the sample-config folder.

Many people are using this functionality already and it is very stable.

Suvir

talkkonnect, 2023-09-09:
[image]

try this but adjust the resistors and supply voltage this is an example for supply of 24 VDC

talkkonnect, 2023-09-11:
Since there is no reply or development on this question I will close this ticket for now.

closed 2023-09-11 by talkkonnect