talkkonnect/talkkonnect #99: Cant build talkkonnect on Ubuntu VM

s3m1s0n1c, 2023-09-10:
I'm trying to build talkkonnect  but get the following error

```talkkonnect@talkkonnect:~/gocode/src/github.com/talkkonnect/talkkonnect$ go build -o /home/talkkonnect/bin/talkkonnect cmd/talkkonnect/main.go
package command-line-arguments
        imports github.com/talkkonnect/talkkonnect
        imports github.com/talkkonnect/max7219
        imports github.com/fulr/spidev: build constraints exclude all Go files in /home/talkkonnect/gocode/src/github.com/fulr/spidev
package command-line-arguments
        imports github.com/talkkonnect/talkkonnect
        imports github.com/talkkonnect/gumble/opus
        imports github.com/talkkonnect/gopus: build constraints exclude all Go files in /home/talkkonnect/gocode/src/github.com/talkkonnect/gopus
```

I'm using version go1.20.3 
```
talkkonnect@talkkonnect:~/gocode$ go version
go version go1.20.3 linux/amd64
talkkonnect@talkkonnect-recorders:~/gocode$ 
```

any ideas?

Thanks Sonic

zrdimetc, 2023-09-10:
https://github.com/talkkonnect/talkkonnect/issues/74
https://github.com/talkkonnect/talkkonnect/issues/59

s3m1s0n1c, 2023-09-10:
Well thats a bit disappointing...

I tried the other version of gopus but couldn't get it to compile..

Oh well

Maybe you should remove the PC stuff then from Readme as its not suited for PC?

Thanks
Sonic

closed 2023-09-10 by s3m1s0n1c

zrdimetc, 2023-09-10:
Majority of users use talkkonnect with single board computer. Just some people, including you and me, need to run talkkonnect in x86-64 platforms sometimes gopus was fathered in from gumble / talkiepi projects and it was working with old Opus 1.1.2. Opus in talkkonnect is version 1.3.1 with numerous quality improvements. It didn't make sense to keep talkkonnect backwards compatible with old gopus / Opus from gumble and talkiepi, just for making rare x86-x64 implementations smoother. People who want this will usually find a way to tweak Opus and build x86-64 talkkonnect's. With new Opus versions 1.4, 1.5 more surprises when building for x86-64 are possible.

s3m1s0n1c, 2023-09-11:
Yeah fair enough.. Makes sense..
Thanks for your help :)