talkkonnect/talkkonnect #100: how to create certificate

andi220886, 2023-09-23:
mumble server requires a certificate to register

talkkonnect, 2023-09-24:
to generate and use a cert with talkkonnect
1. log in via ssh
2.  run /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/scripts/gencert.sh
3. the generated cert will be called mumble.pem and will be copied to your talkkonnect directory
4. edit the talkkonnect.xml file in the account main section and certificate tag add the complete path to the generated certificate like this /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/mumble.pem
5. try it if it works if not then you may have to config the insecure and the register tags in the account section.

Happy talkkonnecting!

talkkonnect, 2023-09-29:
since mo update i will close this ticket

closed 2023-09-29 by talkkonnect