talkkonnect/talkkonnect #101: After updating there are no longer sounds when a user connects or disconnects. other sounds work fine

MrSco, 2023-09-24:
Not sure what happened, but my config looks good and other sounds play ok....

[talkkonnect.xml.txt](https://github.com/talkkonnect/talkkonnect/files/12709106/talkkonnect.xml.txt)
[image]

it doesn't even seem to attempt to play a sound file when a user connects/disconnects. did the event name change or something?

talkkonnect, 2023-09-25:
MrSco,

thanks for your feedback, it was a bug and is fixed already. Update to version 2.31.03 for the event sounds on join and leave channel to work again.

After I implemented the new logic to check for user status I forgot to add back the local media playing of the sounds.

I tested and it works perfectly now.

thank you for your valuable feedback,

suvir kumar

MrSco, 2023-09-25:
ah ok. that fixed the issue. 

in a related note though... it seems the update script has a bug restoring the talkkonnect.xml after update is finished... its looking for github.old folder to copy backup xml file from but it backed it up in the /root folder and so should be copying from there. I end up having to copy back my talkkonnect.xml file from /root manually after update is completed.

thanks for the quick response as always!

talkkonnect, 2023-09-25:
Are you using rasp-talkkonnect script? That's the one to use.

MrSco, 2023-09-25:
from the main github readme, using ...

https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/update-talkkonnect-32bit.sh

I don't see any other update script or anything named rasp-talkkonnect in the repo.

talkkonnect, 2023-09-25:
Sco,
cd /root
use wget https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/raspi-talkkonnect
then chmod +x /root/raspi-talkkonnect
and then run /root/raspi-talkkonnect to run the menu driven script.

MrSco, 2023-09-25:
its ok to use raspi-talkkonnect for an orange-pi zero running armbian?

talkkonnect, 2023-09-25:
I am using it on my orange pi zero running armbian and it works here. So my answer is it should work there too in my opinion.

MrSco, 2023-09-25:
ok. good to know. thanks for the help!

closed 2023-09-25 by MrSco