talkkonnect/talkkonnect #102: Connection and instantaneous disconnection, it's in a loop

f59465-Clem, 2023-10-16:
Hello Suvir,
I made a second relay based as always on Pizero 2w.
The installation with the image from March 3, 2023 went well, but it was impossible to connect to the demo server or to my server.
In fact there is a stealthy connection and instantaneous disconnection, it's in a loop, it constantly repeats 'talkkonnect loader'.

I pinged Google, it works, I have a network.

F59465 Clem

talkkonnect, 2023-10-16:
Hi Clem,
thank as always for your kind feedback and being production user of talkkonnect as a repeater
try to follow the instructions below

1. ps -ax make sure there is only 1 instance of talkkonnect running
1. Update talkkonnect and use version 2.31.04 released 05 OCT 2023
2. modify xml file to set the tags for logging
singleinstance= try to change to false (if it works means there are more than 1 talkkonnect running)
logging=screenandfilewithlineno
loglevel=debug
logfilenameandpath=/var/log/talkkonnect.log
 
Look at the logs, in the file /var/log/talkkonnect.log if they still dont make
any sense send me the logs.

best,
suvir kumar

f59465-Clem, 2023-10-17:
Hello Suvir,
I installed a version of 2022 for pi3b+ no problem it connects directly.
I then used the last image, it was impossible to connect to the server, nothing conclusive on the log, well one instance, I updated this one to connect directly.

So I think there is a problem with the last image, it would be good to update it for future users.
problem solved for me.

Good day
F59465 Clem

talkkonnect, 2023-10-25:
Since no further feedback allow me to close this ticket

closed 2023-10-25 by talkkonnect