talkkonnect/talkkonnect #103: SA818 module wiring

f59465-Clem, 2023-10-23:
Hello everyone,

I am testing the SA818 module for a single RX/TX UHF frequency.
I use PC817 for the PTT pin 5 of the SA818 it works on the RPI GPIO voiceactivity (output).
On the other hand, the COS pin 1 of the SA818, RPI GPIO Txptt (input) gives me a problem, it needs logic 0.

Unless there is another way for Talkkonnect to take and charge the SA818 module, I found no information or diagrams on the wiring.

Thank
F59465 Clem

talkkonnect, 2023-10-26:
Clem,
under the hardware io pins sections of the talkkonnect.xml file you can set  inverted="ftrue" for the pin in which you want the reverse logic to apply.

suvir

f59465-Clem, 2023-10-27:
Suvir,
ok, I didn't know that the "inverted" function also works with GPIO input (COS).
 Good news
Thank

talkkonnect, 2023-10-27:
Clem,
sorry I thought this was for output, I don't think it works for input however let me check again

talkkonnect, 2023-10-27:
Clem,
yes "inverted" only works with output as of now. Let me think of a way to make it work with reverse logic also. But will need a good way to test it. This will take some time to test and fix pending my available time. I will put it on my todo list.

closed 2023-10-29 by talkkonnect