talkkonnect/talkkonnect #104: TX Pin Glitches

s3m1s0n1c, 2023-10-24:
Hey Guys,

We have noticed that when a user’s voice activity is present the TX pin appears to glitch off/on during the voice activity, at kind of regular intervals. 
The glitches are very small initially, but the glitch duration increases greatly after around three minutes of continuous voice activity from the same user, where the glitch duration is very noticeable for a short period, then the whole cycle of events appears to somewhat restart. 
We Have experimented with <voiceactivitytimermsecs>. 
We have noticed that with a setting of 1 the glitches are almost constantly there and a setting of 500 appears to remove all glitches for a continuous voice activity period of several minutes, when testing. We are testing with just a basic TalkKonnect configuration with LED’s. Are there any ramifications related to having increased the setting to 500 using this configuration, or if other features are later activated? 

Thanks Sonic

talkkonnect, 2023-10-26:
Hi Sonic,
if the RX led blinks it means there is some packet loss. For the voiceactivitytimersecs settings is to add a delay in milliseconds to the led so as to "smooth" out these interruptions so that if talkkonnect is being used to key a RF Radio that is sensitive to these dropouts its like a "capacitor" to keep the radio transmitting during these small interruptions. You will have to find the minimum value to set in which not to see the flickering of the LED if you need to drive something with the LED signal.

Is 500 a good value? No idea have to test it with your setup. Are you using it to key a transmitter? Whats your use case?

s3m1s0n1c, 2023-10-26:
Thanks Suvir :)

That's what we thought, it seems 450-500 seems to be the value that works for us..

Yes, Its keying up a simplex radio..

Thanks Sonic

closed 2023-10-26 by s3m1s0n1c