talkkonnect/talkkonnect #106: Oled display not updating while channel change

peterfdej, 2023-10-31:
When you change channel (e.g. F1, F2), the oled display doesnot update de name of the channel. When you press F7 (channel list), or someone is entering the the channel or leave the channel, yet the display refresh the channel name.
I am using image talkkonnect-v2-rpi3-64bit-usbsound.img and an 0.96" oled display.

talkkonnect, 2023-10-31:
Please update to the latest version using update script

talkkonnect, 2023-11-02:
The event system has been changed recently to reflect the status of users much more reliably. For Both sound events and display. Please update and feedback

closed 2023-11-02 by talkkonnect

peterfdej, 2023-11-04:
Finally the update succeeded. Have to run it several times, also started again on a complete new image. Lost the talkkonnect.xml, but now it is updated and up and running.
The Oled display now sometimes update the channel name. When you toggle (e.g. F1 next channel) the display changes when another user is using that channel and it changes when you toggle to the next channel (even when you are the only one on that channel). After that it does not change, until you reach the channel where someone is using that channel.