talkkonnect/talkkonnect #108: Release 16 Nov 2023

f59465-Clem, 2023-11-18:
Good morning,

I updated Talkkonnect to the November 16 version, there seems to be a big problem with the 1.3" OLED display, nothing is displayed correctly, connections, disconnections are no longer displayed, users are no longer able to is no longer displayed.
how to go back to the version before 9 Nov?

PS: there is still no information in the log (info), on the activation of the TXptt pin which is really very problematic.

Thanks
Clem

talkkonnect, 2023-11-19:
Clem,
thanks for your bug report, I have fixed and improved the display messages based on your feedback.
Please update to Ver 2.33.01 Rel 19 Nov 2023 to test and feedback here.

Suvir

talkkonnect, 2023-11-20:
To update to the latest version simply

1. log in via ssh as root
2. cd /root
3. wget https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/update-talkkonnect-64bit.sh
4. chmod +x update-talkkonnect-64bit.sh
5. /root/update-talkkonnect-64bit.sh

talkkonnect, 2023-11-20:
If you want to see tx in the log in the info loglevel.

change line 458 in the program gpio.go (from debug to info)

 log.Println("debug: Tx Button is pressed") ->  log.Println("info: Tx Button is pressed")

then compile talkkonnect again.

f59465-Clem, 2023-11-20:
Hi Suvir,

Thanks for your help and development.
I will update this evening.

I will come back.

f59465-Clem, 2023-11-20:
How do we compile?
Thanks

talkkonnect, 2023-11-22:
to compile talkkonnect use script
1. log in via ssh as root
2. cd /root
3. wget https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/compile
4. chmod +x compile
5. /root/compile

this should compile any changes you make to talkkonnect.

f59465-Clem, 2023-11-22:
Hello Suvir,

I am making a return for oled display, the r2 update. 33.01 corrected the problem, I find the display more responsive, thank.

On the other hand I notice another problem, the formatting of the text does not take into account the underscore (key 8), can you correct this, because the words do not have spaces, all stick to the other,
 moreover Creating authenticated accounts on Mumble does not accept the dash 6.

I'll report back on the compilation if I get there...
greeting

talkkonnect, 2023-11-22:
Ok will add the underscore to the allowed characters on oled

talkkonnect, 2023-11-23:
Clem,
I added underscore into the allowed characters for the Display (OLED)  version "2.34.02" updated "23 Nov 2023"

closed 2023-11-23 by talkkonnect