talkkonnect/talkkonnect #109: Update process 404

f59465-Clem, 2023-11-25:
Hi Suvir,
I see there have been changes to the way Talkkonnect is updated.
It can no longer find the file "update-talkkonnect-64bit.sh"

    1-log in via ssh as root
    2-cd /root
    3-wget https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/update-talkkonnect-64bit.sh
    4-chmod +x update-talkkonnect-64bit.sh
    5-/root/update-talkkonnect-64bit.sh

how do we proceed again?

Sincerely

talkkonnect, 2023-11-26:
Clem,

I have removed the -32 and -64 from the script. Now the script will automatically find out details about the cpu if it is older 32 or newer 64 bit type and install the correct versions of the compiler etc.

1-log in via ssh as root
2-cd /root
3-wget https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/update-talkkonnect.sh
4-chmod +x update-talkkonnect.sh
5-/root/update-talkkonnect.sh

notice the -64bit part missing and for older boards -32bit part is also gone it is now replaced by update-talkkonnect.sh
once script to run them all

suvir

talkkonnect, 2023-11-27:
Updating with the script should work, if any problems re-open this ticket. For now I will consider the issue resolved and close the ticket.

closed 2023-11-27 by talkkonnect