talkkonnect/talkkonnect #110: Release  2.34.03 24Nov

f59465-Clem, 2023-11-27:
Good Suvir ,

I notice a bug or configuration problem.

For the relay I always deactivate all the "sound events" despite this when the relay is in operation and a new user arrives or this disconnects from Mumble the sounds are not played however, this cuts the GPIO voiceactivity 1sec, cut the current user's transmission on the repeater.

This becomes problematic.
Sincerely

talkkonnect, 2023-11-28:
Hi Clem,
when a new user joins there is a drop of audio for a fraction of a second for all clients that is just how the mumble server works. You can test this behavior even with windows official client. All clients behave the same since it is a server side thing and not a client side thing. If the server has a lot of users this becomes very noticeable. 

Please let me know if this is the problem, that a new user joins the server the audio drops for a fraction of a second. For the voice activity you can add to the timer to help for this "flickering of the LED".

Please let me know if this is what you mean in your feedback and I have understood correctly. Also you can verify this with the official client and see if it happens.

thanks,
suvir

f59465-Clem, 2023-11-28:
Hi Suvir,

I sent you the video on messenger.

With each mouse click I connect/disconnect from a Mumble Windows client, the same on a smartphone.
We can clearly see that the GPIO goes from 1 to 0 and the Motorola end of transmission roger beep sounds.

If that can help you.

closed 2023-12-02 by f59465-Clem