talkkonnect/talkkonnect #113: Adjust Mic Gain via MQTT

Keksstar, 2024-01-01:
Hi!

New talkkonnect user here. Realy love the simplicity of it. Great work you did there!

Is there any way to control the Mic Gain via MQTT? Only found output volume, but no input volume.
If there isn´t a way yet, I would like to add this as as an feature request please :-)

Planning to use Talkkonect for remote video production, there I have to deal with different headset types and therefore it would be pretty awesome if I can control all the "user" gains externally.

Thanks!

talkkonnect, 2024-01-02:
Ok thanks for the suggestion. Will look into it when I return home from vacation.

Keksstar, 2024-01-23:
Do you had the time to look into this one? :-)

Keksstar, 2024-02-07:
Is there a way to push this with real money? 😉

talkkonnect, 2024-02-09:
Hi and thank you very much for the offer,
It was always for the challenge and fun. It Was not about the monkey as a driver to create talkkonnect. I am totally swamped with my day job and also so much work in my own company. I will have a look this sunday and see what needs to be done to make this remote volume input control work and get back to you. What kind of hardware are you using and what name does the mic appear in alsamixer?

talkkonnect, 2024-02-14:
The proof of concept is done for mic volume. will need some time to implement it.

Keksstar, 2024-02-14:
Wow! This is awesome news!! Thank you so much for your work.

talkkonnect, 2024-02-18:
Keksstar,
Your feature request for local/remote mic volume control is now available talkkonnect version 2.39.01 released  18 Feb 2024

suvir

Keksstar, 2024-02-18:
Awesome! Will check it out an report back.
Thank you so much for your work!

Keksstar, 2024-02-18:
Where´s the difference between volumerx and volumetx?

Is there a way to get the current Mic Gain value?
Like this one?  <command action="currentvolume" message="Current Volume" enabled="true"/>

talkkonnect, 2024-02-19:
Volumerx is the original received volume adjustment through the speaker and
vokumetx is the transmitted volume which is the microphone volume

Suvir Kumar
Global Coders Co., Ltd.
262/15 Soi Onnut 46,
Onnut SubDistrict, Suan Luang District
Bangkok 10250
THAILAND
Tel. 668 1 840 7160

talkkonnect, 2024-02-20:
I have not built the current volume request yet currently you can remotely adjust the mic volume via volumetx via remote control or local

talkkonnect, 2024-02-22:
Any update? Is it working? Let me know

thanks
suvir

Keksstar, 2024-02-22:
Will check it on the weekend, as I'm quite busy right now.

Currently building some of these, with integrated talkconnect:
[image]
They do 4G uplink, have intercom and camera control (for shading, etc). All buffered with an integrated battery, when no external power is supplied.

talkkonnect, 2024-02-23:
Please send a pic for the blog once completed. So others can be inspired by your work. Will keep this ticket open awaiting your feedback.

suvir

talkkonnect, 2024-02-23:
Keksstar,
all functionality has been added for local and remote mic control. You can get the volume, increase and decrease the volume of the mic now. 

Some terminology 
RX=receive so rxvolume is the received volume in talkkonnect (ie the speaker volume)
TX=transmit so txvolume is the transmitted volume in talkkonnect (ie the microphone volume)

Happy talkkonnecting!

suvir

Keksstar, 2024-02-25:
MQTT testing on hold till #119 is resolved

talkkonnect, 2024-02-27:
OK I will come up with example of how to set up documentation, I will do it when I have some free time from work.

talkkonnect, 2024-02-27:
Dear Kekkstar,
I have updated the Readme.md file in the section for MQTT. 
Link https://github.com/talkkonnect/talkkonnect/blob/main/README.md#the-mqtt-section

Here I have got talkkonnect to subscribe to the topic event/cameraman1/ and when we send the message volumetxup we can increase the camera man 1 microphone volume for example.

It is working I have tested it here. I used a mobile app called mymqtt to publish and test the volumetxup and volumetxdown payloads and talkkonnect is responding to the commands. We can see these responses on talkkonnect logs.

There is no reply for MQTT commands as per design. But this works. You can also use GPIO on one talkkonnect to command another talkkonnect to respond to the button press for example you can press the button locally to command an increase in the remote talkkonnect's mic volume.

Keksstar, 2024-02-27:
Currenttxvolume will respond with an value, right?
Had no success with MQTT Explorer yesterday, will try again.

Thanks!

talkkonnect, 2024-02-27:
No response on the calling mqtt client you will see the response on the
called talkkonnect console log and your will hear the mic volume change on
the receiving side.

Keksstar, 2024-02-27:
But what does this one do then, when not giving an response?

<command action="currentrxvolume" message="Current Volume" enabled="true"/>

talkkonnect, 2024-02-27:
Listen to the camera man's louder or softer voice and rejoice.

Keksstar, 2024-02-27:
Whoops... GitHub did not include the command.
I meant this one:
command action="currentvolume" message="Current Volume" enabled="true"

From the naming I would expect to get an response? Or what does this command do?

Would it be possible for you to add an response to get current Mic and Volume levels via MQTT?

talkkonnect, 2024-02-27:
Try if the effect is seen and you can increase the mic remotely via mqtt or not. If this works for you then we can discuss about reply. What kind of reply you expecting and reply to the topic set in mqttpubtopic right?

talkkonnect, 2024-03-03:
Since no update allow me to close this ticket

closed 2024-03-03 by talkkonnect