talkkonnect/talkkonnect #114: Sidetone?

Keksstar, 2024-01-01:
Is there a way to add sidetone when using Headsets?

I can do it in the Mumble App or the PC Client, but couldn´t find a way on Talkkonnect...

Thanks!

Propeller-v8, 2024-01-02:
Hi there,
you can manage sidetone, also mic gain, input devices, ... by alsa mixer. You can store your preset files to disc.
Suvir can share some knowledge, if MQTT commands can trigger multiple scripts for changing alsa script files... 

Greets Andre

Keksstar, 2024-01-04:
Just need to send it once... haven´t found a way to to this in alsamixer.

talkkonnect, 2024-01-05:
i would recommend to play with the 'Mic Playback Volume' and then do a  do alsactl store 1 so it remembers the setting if you only have to do it once

Keksstar, 2024-01-11:
Well the problem I see here is that I don´t have the loopback device.
I installed an freshley new Armbian with Talkkonnect, but in Alsamixer I don´t have the loopback option.
Also in aplay -l there is only my USB Soundcard listed...
Will try to reinstall with an different distro.

Keksstar, 2024-01-11:
Install of an different Distro (Ubuntu) didn´t helped... I still don´t have the loopback option... :-/

talkkonnect, 2024-01-12:
If by sidetone you mean you can hear yourself in the headset while speaking that is already working its just to set alsa correctly. Do I understand correctly? You want to hear yourself on the headset when you speak?

Keksstar, 2024-01-12:
That's correct.
Tried everything in Alsa :-/

Keksstar, 2024-01-23:
Sorry for the late reply.
I´m using an "NanoPi NEO Core" with Armbian 23.11.1
Not using the premade image, I used your talkconnect install script.

talkkonnect, 2024-01-28:
I can tell you about how to do sitetone for the board I have which is orange pi zero 2H chip. Look Here
https://www.talkkonnect.com/2020/04/21/tips-for-talkkonnect-building-with-orange-pi-zero/

and look at Item 10 titled Sound Settings

here it shows you how to turn the side tone off, do the opposite to turn it on. Beware it can cause feedback and howl really loud.

closed 2024-01-28 by talkkonnect