talkkonnect/talkkonnect #115: multiple server Bis

f59465-Clem, 2024-01-16:
Hello Suvir,

the ctrl+Q key does not work, is there something to do in the talkkonnect.xml file ?

Or even better, I have two servers, I would like to switch from one to the other with an SSH command, this would be easier for a repeater 75km away.

Ps: I was unable to open the previous question close.
Thank
Clem

talkkonnect, 2024-01-17:
HI,
I have mapped the next server now in the version 2.37.02 tp Ctrl-Z instead of Ctrl-Q. This is because somehow Ctrl-Q always has some problems maybe a reserved key combination so it changed now.

Test and let us know if you still have problems.

suvir

f59465-Clem, 2024-01-18:
Hi Suvir,

Ok, I wanted to start with a clean installation with the latest version 2.37.01, 64 for pi3/zero2w.
I still use a Pi zero 2w + Waveshare ETH/USB HAT after copying to the mSD, Talkkonnect does not start, status talkkonnect gives me a red error.

I managed to make it work with the old version 2.19.01
There is something wrong with the latest 2.37.02 in 64 or 32 bits, I tried both.

Greeting
Clem

talkkonnect, 2024-01-19:
Hi Clem,
it works here so i would need the logs to be able to see why its not
working there. Are you using the new image?

suvir

talkkonnect, 2024-01-24:
Since no response further feedback allow me to close this ticket.

closed 2024-01-24 by talkkonnect