talkkonnect/talkkonnect #116: Image  2.37.01 Released 31/December/2023 not work on Rpi Zero 2W

f59465-Clem, 2024-02-17:
Hi Suvir,

Voici le problème avec la derniére image sur mon dernier Rpi Zero 2w.
[image]
hoping to find a solution...

Thank
Clem

talkkonnect, 2024-02-18:
Please help Try ssh in and apt install screen if that works then I will add it to the image and release

f59465-Clem, 2024-02-19:
Hi Suvir,
I installed screen without success...
[image]

Have you successfully installed the 2.37.01 image on a Zero 2W ?
I tried on two Zero 2Ws, it doesn't work.

Thanks
Clem

talkkonnect, 2024-02-19:
I will try when I have time, maybe this weekend. Which Image are you using exactly please post the link.

f59465-Clem, 2024-02-19:
hi,
here: [https://drive.usercontent.google.com/download?id=15--tFwJnKSmWstChioWGrom563KE1vmb&export=download](url)

thank

talkkonnect, 2024-02-19:
OK Thanks I will test it with raspberry pi zero 2w and usb sound card when I have time.

talkkonnect, 2024-02-23:
Hi Clem,
I realized when I set up rapsberry pi zero 2W I will need USB OTG which I do not have. So I cannot test whether USB sound card willl work on rapsberry pi zero2w however the image with respeaker hat works that has been tested.

closed 2024-02-23 by talkkonnect

f59465-Clem, 2024-02-23:
> I will remove the wording RPI Zero 2W USB sound card until I have an OTG
> to test it.

Hi Suvir,
This version not work...

talkkonnect, 2024-02-26:
Clem,
I have made new version for Pi Zero 2W with USB sound Card which is working here, I will release the separate image for pi zero 2w soon.

suvir

talkkonnect, 2024-02-26:
Clem,
new image for RPI Zero 2W with usb OTG cable connected to USB sound card is released and tested that it is working.

f59465-Clem, 2024-02-26:
Thank you Suvir for your responsiveness.

Raspberry Pi Zero 2w, USB OTG, CM108, USB 4g LTE, for me it is the perfect combo for RF Gatway repeater.

I'll try that, I'll report back.

F59465 Clem

f59465-Clem, 2024-02-26:
> Clem, new image for RPI Zero 2W with usb OTG cable connected to USB sound card is released and tested that it is working.

Suvir, is it possible to have it in 64 Bit version? 
I find it a shame to restrict the Zero 2w which performs very well in 64Bit

Thank
F59465 Clem

talkkonnect, 2024-02-27:
> Hi Suvir,
> invalid link for RPI Zero 2W, OTG, 64bit.
>
> <
> https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
>
> Sans
> virus.www.avast.com
> <
> https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
>
> <#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Dear Clem,
try to download again pls. the version is 64 bit already i copied and
pasted that is 32 bit by mistake and forgot to change the wording to 64 bit.
let me know your results.

best regards
suvir

f59465-Clem, 2024-02-27:
Hi Suvir,

I tried the RPI Zero 2w, OTG, OLED, CM108 version.
For me it doesn't work, see screenshot
I tried the default config on "raspberry pi Imager" only ssh active
I also tried the personal config "hostname,ssh,time zone, keyboard type Fr.
The oled screen also does not work, despite "oled enabled=true" in the xml.
I notice that the "talkkonnect" folder is missing, to modify the xml,
I have to do /home/pimum2/gocode/src/github.com/talkkonnect/talkkonnect/talkkonnect.xml
[image]

thanks
F59465 Clem

talkkonnect, 2024-02-27:
Doesn’t make any sense cause it’s working here perfectly. I even burned the image to a new card, it booted up and connected to WiFi out of the box. What did you setup with the imager exactly?

talkkonnect, 2024-02-27:
What is this pimum folder?

f59465-Clem, 2024-02-27:
pimum2 or 1-3-4 is the user name, ssh, the machine name, different on each machine.

Yes I am connected to the internet, there is a ping at the start of the screenshot
[image]

thank
F59465 Clem

talkkonnect, 2024-02-27:
Ssh is already active in my image no need to do all this when you get the image insert written sd into your windows machine and create wpa-supplicant that’s all don’t modify using pi imaging tools.

f59465-Clem, 2024-02-27:
I don't use talkkonnect ssh, I disable it, I need full access to the machine with the highest protection with EdDSA keys.
I don't use Wifi, I've always been connected via LAN with the W5500 ethernet module in SPI plug and play on Raspbian, it works very well because I'm connected to the internet.

I'm going to try with SSH from Talkkonnect, I have doubts that it comes from there.

Thank
F59465 Clem

f59465-Clem, 2024-02-27:
Suvir,
I see in the description of the image, that we must be on LAN connected to the box, is there something to modify in the config.txt file?

Thank
F59465
Clem

talkkonnect, 2024-02-27:
The easiest thing I would recommend is get the image burn it to SD card
with no modifications. Connect a micro USB cable and power the raspberry pi
zero 2 w to your PC powering from PC. Then use putty over com port to log
into the pi. Using user root and password talkkonnect. Then use rasp-config
tool to connect to wifi. If it still doesn't work then I am out of ideas.

f59465-Clem, 2024-02-27:
Hi Suvir,

Rpi Pi Imager 1.8.5 latest release
Default classic user configurator "Pi"

I did some tests:
As said previously, he cannot find the "talkkonnect" folder in "home" after searching it does not exist.
It puts "bin and gocode" in the "pi" folder instead of putting it in "home/talkkonnect"

May be a way for you to resolve the problem

Thank
F59465 Clem