talkkonnect/talkkonnect #117: tkbuild.sh fails with go module error

lkb2k, 2024-02-21:
Hi there,  I'm trying to get up and running with the installer script on a Raspberry Pi 4 with a waveshield audio hat.  Running the tkbuild.sh script results in this error:

```
go: go.mod file not found in current directory or any parent directory.
	'go get' is no longer supported outside a module.
	To build and install a command, use 'go install' with a version,
	like 'go install example.com/cmd@latest'
	For more information, see https://golang.org/doc/go-get-install-deprecation
	or run 'go help get' or 'go help install'.
./tkbuild.sh: line 85: cd: /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect: No such file or directory
```

The line that is failing is 
```
go get -v github.com/talkkonnect/talkkonnect
go: go.mod file not found in current directory or any parent directory.
	'go get' is no longer supported outside a module.
	To build and install a command, use 'go install' with a version,
	like 'go install example.com/cmd@latest'
	For more information, see https://golang.org/doc/go-get-install-deprecation
	or run 'go help get' or 'go help install'.
```

I'm not super familiar with the go build chain,  do I need to use 'go install' instead of 'go get' somehow?

talkkonnect, 2024-02-21:
There are some changes in the build system that breaks go get on version golang 1.22. I have already fixed the update-talkkonnect.sh script next I will fix the tkbuild.sh and release it soon.

talkkonnect, 2024-02-21:
For golang version 1.22 and above I have fixed the tkbuild.sh script already. It's released in github talkkonnect/scripts directory.
It is available at https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/tkbuild.sh

Please use this script instead to build talkkonnect and let us know your results, thank you for your feedback.

Suvir

lkb2k, 2024-02-21:
```
installing talkkonnect with traditional method avoiding go get cause its changed in golang 1.22
mkdir: cannot create directory ‘/home/talkkonnect/gocode/src/github.com/talkkonnect’: No such file or directory
./tkbuild.sh: line 86: cd: /home/talkkonnect/gocode/src/github.com/talkkonnect: No such file or directory
Cloning into 'talkkonnect'...
remote: Enumerating objects: 5328, done.
remote: Counting objects: 100% (983/983), done.
remote: Compressing objects: 100% (332/332), done.
remote: Total 5328 (delta 734), reused 867 (delta 648), pack-reused 4345
Receiving objects: 100% (5328/5328), 40.68 MiB | 4.49 MiB/s, done.
Resolving deltas: 100% (2999/2999), done.
./tkbuild.sh: line 88: cd: /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect: No such file or directory
go: cannot determine module path for source directory /home/talkkonnect/gocode (outside GOPATH, module path must be specified)

Example usage:
	'go mod init example.com/m' to initialize a v0 or v1 module
	'go mod init example.com/m/v2' to initialize a v2 module
```

I updated this line
```
mkdir /home/talkkonnect/gocode/src/github.com/talkkonnect
```

to 
```
mkdir -p /home/talkkonnect/gocode/src/github.com/talkkonnect
```

which fixed it for me

talkkonnect, 2024-02-22:
Fixed as per your feedback and modified script on github.com
Thanks for your kind feedback,

Suvir Kumar

closed 2024-02-22 by talkkonnect

Keksstar, 2024-02-25:
Please update the install script accordingly. mkdir -p change is not in there right now for the talkkonnect directory

talkkonnect, 2024-02-25:
Ok sorry check again pls