talkkonnect/talkkonnect #118: Bug Rotary Encodeur  KY-040

f59465-Clem, 2024-02-21:
Hi Suvir,

On the last update, changing 'mumblechannel' with the rotary encoder no longer works, Talkkonnect disconnects if you turn the knob.

Thank
Clem

talkkonnect, 2024-02-22:
It works but have to see why it disconnects, cause of permissions or what there  must be an underlying cause and that is server dependent, I need the logs to see whats happening. It's a case of it's not happening here so with this limited information I cannot determine the cause of the problem. Please provide logs. You will have to setup logging to file and send me the part where it disconnects.

f59465-Clem, 2024-02-22:
[image]

talkkonnect, 2024-02-23:
The logs have to show when it crashes when you turn the rotary knob.this is showin normal start up of talkkonnect

f59465-Clem, 2024-02-23:
a little more complete
[image]

talkkonnect, 2024-02-23:
It shows rotation and roll over let me test here again. I will get back to you again and also post what my logs look like. Thanks for your feedback

f59465-Clem, 2024-02-23:
yes that's it when I turn clockwise, it displays "rotating Clockwise" no action expected, no change of channel.

when I turn counterclockwise Talkkonnect restarts...

ok, thank

talkkonnect, 2024-02-23:
For me here its working, please send me your talkkonnect.xml file by email. Below are what my logs look like. I am on version 2.39.01 released Feb 2024 same as the version on your logs.

example logs below that are working.

 debug ] 2024/02/23 08:04:47 gpio.go:784: Rotary Button is pressed                                                        
[  info ] 2024/02/23 08:04:47 gpio.go:1599: Current Rotary Item 1 Function mumblechannel                                   
[ debug ] 2024/02/23 08:04:47 display.go:129: Max7219 Seven Segment Not Enabled                                            
[ debug ] 2024/02/23 08:04:47 gpio.go:782: Rotary Button is released                                                       
[ debug ] 2024/02/23 08:04:51 gpio.go:1534: Rotating Clockwise                                                             
[  info ] 2024/02/23 08:04:51 clientcommands.go:284: Moving to Accessable Channel (ID 30 Name HS2AN-NYK)                   
[ debug ] 2024/02/23 08:04:51 gpio.go:1534: Rotating Clockwise                                                             
[  info ] 2024/02/23 08:04:51 clientcommands.go:284: Moving to Accessable Channel (ID 33 Name Samchuk340HPV)
[ debug ] 2024/02/23 08:04:51 gpio.go:1301: Turning On participants at pin 6 Output GPIO (Non-Inverting)                   
[ debug ] 2024/02/23 08:04:51 gpio.go:1331: Turning Off participants at pin 6 Output GPIO (Non-Inverting)                  
[ debug ] 2024/02/23 08:04:53 gpio.go:1534: Rotating Clockwise                                                             
[  info ] 2024/02/23 08:04:53 clientcommands.go:284: Moving to Accessable Channel (ID 48 Name Mercedes Benz 01)
[ debug ] 2024/02/23 08:04:53 gpio.go:1534: Rotating Clockwise                                                             
[  info ] 2024/02/23 08:04:53 clientcommands.go:284: Moving to Accessable Channel (ID 53 Name talKKonnect)
[ debug ] 2024/02/23 08:04:53 gpio.go:1301: Turning On participants at pin 6 Output GPIO (Non-Inverting)                   
[ debug ] 2024/02/23 08:04:54 gpio.go:1331: Turning Off participants at pin 6 Output GPIO (Non-Inverting)                  
[ debug ] 2024/02/23 08:04:56 gpio.go:1559: Rotating CounterClockwise                                                      
[  info ] 2024/02/23 08:04:56 clientcommands.go:310: Moving to Accessable Channel (ID 48 Name Mercedes Benz 01)
[ debug ] 2024/02/23 08:04:56 gpio.go:1559: Rotating CounterClockwise                                                      
[  info ] 2024/02/23 08:04:56 clientcommands.go:310: Moving to Accessable Channel (ID 48 Name Mercedes Benz 01)
[ debug ] 2024/02/23 08:04:56 gpio.go:1301: Turning On participants at pin 6 Output GPIO (Non-Inverting)                   
[ debug ] 2024/02/23 08:04:57 gpio.go:1559: Rotating CounterClockwise                                                      
[  info ] 2024/02/23 08:04:57 clientcommands.go:310: Moving to Accessable Channel (ID 33 Name Samchuk340HPV)
[ debug ] 2024/02/23 08:04:57 gpio.go:1559: Rotating CounterClockwise                                                      
[  info ] 2024/02/23 08:04:57 clientcommands.go:310: Moving to Accessable Channel (ID 33 Name Samchuk340HPV)
[ debug ] 2024/02/23 08:04:57 gpio.go:1331: Turning Off participants at pin 6 Output GPIO (Non-Inverting)                  
[ debug ] 2024/02/23 08:04:59 gpio.go:1559: Rotating CounterClockwise                                                      
[  info ] 2024/02/23 08:04:59 clientcommands.go:310: Moving to Accessable Channel (ID 30 Name HS2AN-NYK)                   
[ debug ] 2024/02/23 08:04:59 gpio.go:1534: Rotating Clockwise                                                             
[  info ] 2024/02/23 08:04:59 clientcommands.go:284: Moving to Accessable Channel (ID 48 Name Mercedes Benz 01)
[ debug ] 2024/02/23 08:04:59 gpio.go:1301: Turning On participants at pin 6 Output GPIO (Non-Inverting)                   
[ debug ] 2024/02/23 08:05:00 gpio.go:1559: Rotating CounterClockwise                                                      
[  info ] 2024/02/23 08:05:00 clientcommands.go:310: Moving to Accessable Channel (ID 33 Name Samchuk340HPV)
[ debug ] 2024/02/23 08:05:00 gpio.go:1559: Rotating CounterClockwise                                                      
[  info ] 2024/02/23 08:05:00 clientcommands.go:310: Moving to Accessable Channel (ID 33 Name Samchuk340HPV)
[ debug ] 2024/02/23 08:05:00 gpio.go:1331: Turning Off participants at pin 6 Output GPIO (Non-Inverting)

f59465-Clem, 2024-02-23:
I sent to your email

talkkonnect, 2024-02-23:
I have replied your email with working xml file seems the config file was the problem. Email below for you kind reference

Dear Clem,I have modified the xml file and it now works on my server since I dont have the password for your server I tested on the talkkect server.I have attached the file here
What I modified1. the server connection details and user name to connect to my server2. turn off heartbeat <pin direction="output" device="led/relay"     name="heartbeat" pinno="17" type="gpio" chipid="0" inverted="false" enabled="false"/>because that pin on my hardware was mapped to the rotary 
3. modified the rotaryencoder section to match my pins<pin direction="input"  device="rotaryencoder" name="rotarya" pinno="17" type="gpio" chipid="0" enabled="true"/>
          <pin direction="input"  device="rotaryencoder" name="rotaryb" pinno="18" type="gpio" chipid="0" enabled="true"/>
          <pin direction="input"  device="rotaryencoder" name="rotarybutton" pinno="27" type="gpio" chipid="0" enabled="false"/>
Also NOTICE! there for the item rotaryencoder to map the rotary button it should be rotarybutton as shown and not only the word button.
this config works and can change channel and push rotary button.
so Its the XML file tag changes i Guess that was causing the problem. Start with this xml file attached and work your way slowly to change the gpio to match your boardand work on my server to make sure it works first then put your server later.
best,suvir

closed 2024-02-23 by talkkonnect