talkkonnect/talkkonnect #119: Rotary Encoder can´t get original volume?

Keksstar, 2024-02-25:
Enabled the rotary encoder to change volume but it´s not working... Found this in the screen session:

[  info ] 2024/02/25 16:05:42 commandkeys.go:225: F6 pressed Volume Down (-)
[  warn ] 2024/02/25 16:05:42 commandkeys.go:228: unable to get original volume: failed to execute "pactl list sinks" (exit status 1) volume control will not work!

I´m using an USB Soundcard. Volume is labeled as "Speaker" in alsamixer.

talkkonnect, 2024-02-25:
For the latest build I used Golang 1.22 also I found some versions of the operating system comes with pulse audio installed. Check if pulse audio is imstalled. If it is installed You will have to remove pulse audio and get alsamixer working without pulse audio.

Keksstar, 2024-02-25:
I removed pulseaudio, alsamixer is working - I have sound In/Out. But the error is still there.

pactl fails without pulseaudio?

talkkonnect, 2024-02-25:
Are you using the lastest version of the volume-go library? Is this a new install or is it built on a board that previously had talkkonnect installed with the older libraries?

Keksstar, 2024-02-25:
Freshly new installed via your script. Wiped the whole card, installed Armbian on it, used your script.
Maybe Armbian is the wrong distro for that? Should I try something else?
Board is an NanoPi Neo Core with an generic USB Soundcard.

talkkonnect, 2024-02-26:
Ok Armbian may be different the issue is volume library still finds the command pactl in the users search path. Install locate and find the pactl file and delete it.

apt install locate
updatedb
locate pactl

If you see it please delete it. Then try again.

Keksstar, 2024-02-26:
Suvir, you´re the BEST! This works absolutly fantastic.

pactl was in /user/bin/pactl
Removed it and now it works.

closed 2024-02-26 by Keksstar