talkkonnect/talkkonnect #120: Can not connect to own Mumble server

lkb2k, 2024-02-26:
I'm running into issues getting talkkonnect to communicate with my own mumble server.  I can connect to mumble.talkkonnect.com and I can connect to my local LAN mumble server with the official mumble client on MacOS or iOS, but when I try to connect talkkonnect to my mumble server, I get Ping failures and then it disconnects.

```
Ping Server Error read udp 192.168.7.92:33855->192.168.7.217:64738: i/o timeout try 1
```

Any suggestions on how to troubleshoot?  Is there a specific version of the mumble server I should be using?

talkkonnect, 2024-02-27:
Very difficult to troubleshoot a server that I have no idea about the server or its network setup. 

On the community mumble server I  used mumble 1.6 compiled, but the version from debian apt get also work reliably. 

Can you try with other clients like mumble for windows, and mumla for android under the same internet conditions.
Do those clients connect successfully? Is it a talkkonnect specific problem or do other clients also not work?

If other clients also have the same problem I would look if there is a firewall blocking something.

If it is a talkkonnect specific problem I would look if the server requires certificates and generate them accordingly.

suvir

lkb2k, 2024-02-28:
I don't have an android device, but I've confirmed it works fine with the windows, Mac and iOS clients.  I've tested TCP and UDP connectivity with netcat, and the raspberry pi and the server can definitely communicate on those ports.

A certificate issue sounds promising, is there any way for me to get more detailed output from talkkonnect on the failure?  

-Lkb

talkkonnect, 2024-02-28:
You can turn on logging to log to file in the talkkonnect.xml file set the logging to screenandfilewithlineno and then see the logs as set in /var/log/talkkonnect.xml and see the failure message. You can use the gencert.sh script to generate a script and then add the path to talkkonnect.xml if the error is due to certificate. If it is in LAN security should not be such an issue perhaps you can adjust the mumur ini file to relax the security.

closed 2024-02-28 by lkb2k