talkkonnect/talkkonnect #121: My return release 2.40.01 Feb 2024

f59465-Clem, 2024-03-02:
Hi to all,

Test 03/02/2024, on Rpi Zero 2, CM108, OLED, LAN.

1-the default User name and "root" if we change to another user it no longer works, it would be good for it to work when we create another "User name" in ssh on 'pi imager' like on the old ones version of Talkkonnect, there are fewer problems when you just update from an old version.

2- the OLED screen does not work, despite true in the .xml, I managed to get the color by installing i2c-tools, the whole screen is on, but no text.

3-Many "Main Menu" shortcuts do not work or they display, key not mapped, or do two things at the same time.

4-Apparently the administration part, channel moving, mute, unmute, etc., only works Murmur 1.6, while the latest version of the official repository is 1.3, it is in fact this version that 80-90% of people use/install on linux, it would be great to restore this functionality for 1.3, it is better to wait for a release which fixes this, because we no longer have administration possible in the event of a problem.

Thank
F59465 Clem

talkkonnect, 2024-03-04:
Answers to your feedback
There are many changes to the OS version, Pi Imager, GoLang and talkkonnect. Using the latest we can get features like channel listening etc.

1. There are some changes in pi imager and for some reason cannot make modifications to the user with the pi imager anymore if it is a non-standard image with no root. The work around is to use pi  imager to burn the sd card then use the method of placing a wpa_supplicant.conf file,
 
2. the oled screen works here, please check if i2c is enabled and check with i2c tools that you can indeed see the oled and also make sure you are not using gpio 2 and gpio 3 for other generic gpio in the talkkonnect.xml file

3. This main menu and other keys there are some issues I agree

4. If you want to use older version of murmur server there is older version of talkkonnect that works with the older server version, so I suggest you use the version matching each other. The debian repository for the server is very old and many good new features including audio enhancements have been added to the later versions of murmur server.

talkkonnect, 2024-03-04:
As already a user familiar with talkkonnect and raspberry pi another alternative is install the OS yourself and using the install script to install talkkonnect.

talkkonnect, 2024-03-05:
Clem see the new readme of how to write the image using the pi imager so it doesnt remove the talkkonnect user.

when writing dowloaded talkkonnect image using raspberry pi imager do the following

    Click Choose Device (select No Filtering)
    Click Choose OS (Use Custom) choose the talkkonnect img downloaded
    Choose Storage
    Click Next
    Would you like to apply OS customization settings? NO!!!!!
    All existing data on Mass Storage Device will be Erased. Yes

Follow these steps please don;t customize the OS and claim that the talkkonnect user has be deleted since you customized the OS with the raspberry pi images which over writes all existing users.

f59465-Clem, 2024-03-05:
Thank you Suvir,

I'm waiting for a new mSD and I'll give it a try.
I will come feedback.

Thank
F59465 Clem

talkkonnect, 2024-03-10:
since no more activity here allow me to close this ticket