talkkonnect/talkkonnect #122: Issues during installation

besi, 2024-03-04:
Hello there

I did want to give talkkonnect a shot since it seemed to be the only alternative based on dchote's talkiepi that is still under active development.

I have a raspberry pi zero and a re-speaker so I did download the image at 

https://drive.usercontent.google.com/download?id=1XPuzVnlYxRnatF4EvBT9r1aME9zSTKHg&export=download
    ->  talkkonnect-v23701-rpi3-respeaker.img

In the readme it stated that the service would already be running but the folder /home/talkkonnect was empty.
I had a look around and found `root/tkbuild.sh` which I executed and in the process `home/talkkonnect` was created.

However there were some errors along the way and the binaries in `home/talkkonnect/bin` weren't installed neither was `/root/talkkonnect-run` which was referred to in the Readme of this repository.

## Errors during `tkbuild.sh`

> --2024-03-04 12:31:34--  http://go1.22.0.linux-armv6l.tar.gz/
> Resolving go1.22.0.linux-armv6l.tar.gz (go1.22.0.linux-armv6l.tar.gz)... failed: Name or service not known.
> wget: unable to resolve host address ‘go1.22.0.linux-armv6l.tar.gz’

> go/
> go/CONTRIBUTING.md
> go/LICENSE
> go/PATENTS
> go/README.md
> go/SECURITY.md
> go/VERSION
> go/api/
> ....
> ...
> ...
> very
> long
> list
> of
> files
> ...
> go/test/used.go
> go/test/utf.go
> go/test/varerr.go
> go/test/varinit.go
> go/test/winbatch.go
> go/test/writebarrier.go
> go/test/zerodivide.go
> go/test/zerosize.go
> go: go.mod file not found in current directory or any parent directory.
> 	'go get' is no longer supported outside a module.
> 	To build and install a command, use 'go install' with a version,
> 	like 'go install example.com/cmd@latest'
> 	For more information, see https://golang.org/doc/go-get-install-deprecation
> 	or run 'go help get' or 'go help install'.
> ./tkbuild.sh: line 85: cd: /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect: No such file or directory
> cannot find package "cmd/talkkonnect/main.go" in any of:
> 	/usr/local/go/src/cmd/talkkonnect/main.go (from $GOROOT)
> 	/home/talkkonnect/gocode/src/cmd/talkkonnect/main.go (from $GOPATH)
> => Finished building TalKKonnect
> => talkkonnect binary is in /home/talkkonect/bin (that's a lie =)) 
> => Now enter Mumble server connectivity details
> talkkonnect.xml from /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect
> and configure talkkonnect features. Happy talkkonnecting!!

I'm not very familiar with go so I can't really debug this issue myself. 
Any help would be appreciated.

Kind regards

talkkonnect, 2024-03-05:
Besi,
when writing dowloaded talkkonnect image do the following
1. Click Choose Device (select No Filtering)
2. Click Choose OS (Use Custom) choose the talkkonnect img downloaded
3. Choose Storage
4. Click Next
5. Would you like to apply OS customization settings? NO!!!!!
6. All existing data on Mass Storage Device will be Erased. Yes

Follow these steps please don;t customize the OS and claim that the talkkonnect user has be deleted since you customized the OS with the raspberry pi images which over writes all existing users.

try and let me know,
suvir

besi, 2024-03-05:
Ok thanks very much it worked like a charm. 
Please note that `/root/talkkonnect-run` is still not present but `/home/talkkonnect/bin/talkkonnect` is there. 
But maybe the Readme is outdated in that respect (It's a big Readme after all =) ).

closed 2024-03-05 by besi

talkkonnect, 2024-03-05:
Thank you based on your feedback I have fixed the readme and added the correct steps of how to use the raspberry pi imager with the provided images in the readme.

I was wondering why people are saying the talkkonnect user is not there, it is because they were trying to customize the users using the raspberry pi imager which will simply overwrite the users and files created in the image rendering the image useless.

besi, 2024-03-05:
I agree it's a mean issue. Since most of my Pi's are headless I do like to at least add one wifi so I can later add or update the necessary `wpa_supplicant.conf`. Also it's nice to set the hostname upfront so I don't have ten anonymous raspberrypi floating around.

besi, 2024-03-05:
> I was wondering why people are saying the talkkonnect user is not there, it is because they were trying to customize the users using the raspberry pi imager which will simply overwrite the users and files created in the image rendering the image useless.

I consider this a bug of the Imager

talkkonnect, 2024-03-05:
the thing is it was working in the older versions of the imager you could make some changes, now the imager simply overwrites all the user directory. Can't really call it a bug the imager feature where you could customize the OS was made for clean images loaded from the raspberry pi guys, our image is modified and is not stock so the expected behavior of rasperry pi imager my be different.

besi, 2024-03-05:
I filed an issue raspberrypi/rpi-imager#831