talkkonnect/talkkonnect #123: Transmit LED stays on 2-3 seconds after release and Rogerbeep is triggered

besi, 2024-03-05:
I have two RapiZero with Respeakers and when I briefly press the TX button and speak briefly the LED stays on for around 3 seconds before it goes out.

Log

```
 debug ] 2024/03/05 17:48:05 gpio.go:458: Tx Button is pressed                                                                     
[  info ] 2024/03/05 17:48:05 clientcommands.go:105: Speaker Muted                                                                  
[ debug ] 2024/03/05 17:48:05 gpio.go:1503: Turning On LEDStrip Transmit LED                                                        
[ debug ] 2024/03/05 17:48:06 gpio.go:1511: Turning Off LEDStrip Transmit LED         
Here I stop talking
                                              
[ debug ] 2024/03/05 17:48:06 gpio.go:1503: Turning On LEDStrip Transmit LED                                                        
[ debug ] 2024/03/05 17:48:06 stream.go:128: Rogerbeep Playing                                                                      
[  info ] 2024/03/05 17:48:06 stream.go:281: File .../rogerbeeps/Waterdrop.wav Playing!
[ debug ] 2024/03/05 17:48:09 gpio.go:1511: Turning Off LEDStrip Transmit LED                                                       
[  info ] 2024/03/05 17:48:09 clientcommands.go:168: Speaker UnMuted                                                                
```

besi, 2024-03-05:
I was able to resolve the issue by disabling the following line (`enabled="false"`) in my config:

            <sound event="rogerbeep" file="/home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/soundfiles/rogerbeeps/Waterdrop.wav" volume="50" blocking="false" enabled="false"/>

f59465-Clem, 2024-03-05:
yes the roger beep has always had a bug, it beeps 2-3 seconds after hanging up PTT.

talkkonnect, 2024-03-05:
There is this xml parameter called voiceactivityyimermsecs that will keep the led on for an extended period for those slower RF transceivers I would adjust this value to zero if you expect immediate response from voiceactivity led

closed 2024-03-05 by besi