talkkonnect/talkkonnect #125: Workflow to set up the premade images?

Alaric84, 2024-04-04:
Hey,
thanks for providing some premade images.
Do you have any workflow, documentation or even sth like an ansible playbook how you create them? Would be great to know/see what has been done/changed.

Thanks again
Andy

talkkonnect, 2024-04-05:
You can see the scripts directory for installing and upgrading talkkonnect. However for special devices like the respeaker hat you will have to first compile the kernelmdriver for the respeaker. The instructions for compiling the kernel for respeaker you can use the script provided by seed studio or also look at the script provided by adafruit.

Alaric84, 2024-04-08:
ok - got it: There is quite some commented code thats required for the led stripe.
Now it works fine!
Isn't there an easier way to disable the code sections (I think you want to avoid the dependencies?)

Maybe using go build tags?

talkkonnect, 2024-04-08:
Thanks for the feedback I never knew about build tags before. I am also learning new things all the time. Not a professional programmer so lots to learn. With build tags I will be able to modularize talkkonnect. 

Thanks for the recommendation and hope you enjoy using talkkonnect.

talkkonnect, 2024-04-09:
will look into build tags to modularize talkkonnect, the reason I chose to comment it out was the library periph.io was giving a lot of users that were not even using the respeaker trouble compiling the code as the were changes that broke periph.io too many times that I chose to comment it out.

Anyway go tags is very interesting I will have to learn about it.

suvir

closed 2024-04-09 by talkkonnect