talkkonnect/talkkonnect #126: [Solution] Vox Script with a debouce

MarceloAri, 2024-04-07:
Based on the script found here: https://github.com/talkkonnect/talkkonnect/issues/4 
I implemented improvements in order to reduce accidental activations of the tx ptt pin due to a bouce effect.
I needed this solution to make a Vertex Standard VX-2200 radio work using talkkonnect as a RoIP gateway
The script works well, however due to the loop that reads if the audio card is reproducing something, a core is always in 100% use.
I tried to implement the time.sleep() function, but it didn't work well, as the code read the audio data once, but the next time it failed and printed the signal as zero.
If anyone is interested and knows a way to reduce these readings, or a way to implement the time.sleep() function that does not generate reading failures, your contribution is welcome!
I'm not a programmer, but I've studied programming logic, and I've programmed in C.
Eu sou do Brasil!
Thanks.
-------------------------------------------script-------------------------------------------------------
#!/usr/bin/env python
#coding: utf8

## This is a VOX-Script.
## Modified by Marcelo Ari Wiebbelling
## Add software debouce

level = 1000                ## Define level limit to detect audio
marker = 0
count = 0
count_limit = 50            ## Define cout limit to activate tx pin
count2 = 0

import alsaaudio, time, audioop
import OPi.GPIO as GPIO
import sys

inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE,alsaaudio.PCM_NONBLOCK,
    channels = 1,
    rate = 8000,
    format = alsaaudio.PCM_FORMAT_S16_LE,
    periodsize = 160)

try:
    while True:
        l,data = inp.read()
        if l:        
            try:
                wert = audioop.max(data, 2)
                #print(wert)
            except:
                wert = level + 10
               
            if wert > level:
                count = 0
                count2 = count2 + 1 
                if marker == 0 and count2 > 10:
                    print("TX On")
                    file = open ("/sys/class/gpio/gpio2/device/gpio/gpio2/active_low","w")
                    file.write("0")
                    file.close()
                    marker = 1
                    
                    
            else:
                #print("desligado", wert)
                count = count + 1
                count2 = 0 
                if marker == 1 and count > count_limit:
                    print("TX Off")
                    file = open ("/sys/class/gpio/gpio2/device/gpio/gpio2/active_low","w")
                    file.write("1")
                    file.close()
                    marker = 0
                        

except (KeyboardInterrupt, SystemExit):
    file = open ("/sys/class/gpio/gpio2/device/gpio/gpio2/active_low","w")
    file.write("1")
    file.close()
    print("Finish")
    
sys.exit()
----------------------------------------------finish----------------------------------------------------
[vox_orig.zip](https://github.com/talkkonnect/talkkonnect/files/14898001/vox_orig.zip)

talkkonnect, 2024-04-24:
Hi MarceloAri,
Since I am not good with python, I will keep this ticket open for another month if no takers or contributors I will close it.

talkkonnect, 2024-05-22:
Since there are no takers I will close this ticket.

closed 2024-05-22 by talkkonnect

zrdimetc, 2025-06-03:
Here is a quick fix from python gurus from our talkkonnect community...

Cause of a problem is a "busy-wait" loop in the program, constantly checking for new audio data, even when there is none - instead of just waiting efficiently without eating up CPU resources.

`time.sleep()` with `alsaaudio.PCM_NONBLOCK` cause more issues. This is because audio buffer might just not be ready after an arbitrary sleep. And this could lead to missed data or zero reads.

More effective way to handle vox logic with alsa audio is switching from "non-blocking" mode to "blocking" mode and using `alsaaudio.PCM_BLOCK`.

By setting `alsaaudio.PCM_BLOCK`, the `inp.read()` function pauses execution until a full period size of audio data is available from the sound card. This way a vox script isn't constantly checking. Instead, it's waiting for the audio hardware to provide the data. This reduces a CPU usage, because the process returns control to the operating system during the waiting.

```
import alsaaudio, time, audioop
import OPi.GPIO as GPIO
import sys

level = 1000
trashhold = 1 # This variable is not used in the current script
marker = 0
count = 0
count_limit = 50
count2 = 0

# Change PCM_NONBLOCK to PCM_BLOCK
inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_BLOCK, # <--- CHANGE THIS LINE
    channels = 1,
    rate = 8000,
    format = alsaaudio.PCM_FORMAT_S16_LE,
    periodsize = 160)

try:
    while True:
        # This read call will now block until 160 samples are available (20ms at 8kHz)
        l,data = inp.read()
        if l:        
            try:
                wert = audioop.max(data, 2)
                #print(wert)
            except:
                # Fallback if audioop.max fails (e.g., empty data, though less likely with blocking mode)
                wert = level + 10 
               
            if wert > level:
                count = 0
                count2 = count2 + 1 
                if marker == 0 and count2 > 10:
                    print("TX Ativado")
                    # Ensure the path is correct for your OPi GPIO setup
                    file = open ("/sys/class/gpio/gpio2/device/gpio/gpio2/active_low","w")
                    file.write("0")
                    file.close()
                    marker = 1
                    
            else:
                #print("desligado", wert)
                count = count + 1
                count2 = 0 
                if marker == 1 and count > count_limit:
                    print("TX Desligado")
                    # Ensure the path is correct for your OPi GPIO setup
                    file = open ("/sys/class/gpio/gpio2/device/gpio/gpio2/active_low","w")
                    file.write("1")
                    file.close()
                    marker = 0
                        

except (KeyboardInterrupt, SystemExit):
    # Ensure TX is turned off on exit
    file = open ("/sys/class/gpio/gpio2/device/gpio/gpio2/active_low","w")
    file.write("1")
    file.close()
    print("encerrou")
    
sys.exit()
```

Talkkonnect vox functionality is under testing right now. Please check for updates about this feature in the comming weeks.