talkkonnect/talkkonnect #129: Updating Kernel / World

Pegasus454, 2024-06-03:
Hi Everyone, 

  As so many people before me have said, I would also like to extend a thank you to everyone for putting this project together.  I am in the process of spinning a PCB to interface with some of my Kenwood repeaters and link them over the internet. 

  I was wondering, I have used the prebuilt img file for the RPI4b and updated the software. This was all working OK (once I repaced the update script and rebuilt) but when I did an "apt update" and then "apt upgrade" all manners of bad started to happen! 

  It looks like I may have broken the GPIO support or maybe even the CM108 audio device. I will have to wait until I get home to look at what happened, but I was wondering if there was a guide or a recommendation on how to update the kernel and world from the pre-installed image? 

Kind Regards,
Peg

talkkonnect, 2024-06-04:
Hi Peg,
There are scripts of how to update talkkonnect and install talkkonnect. However since there are so many variants of hardware, drivers and operating systems out there that can run talkkonnect.

The actual recommended way is for you to install your own OS and drivers then run tkbuild.sh to setup talkkonnect.
This can be difficult for beginners so as a result I put of the images that I created for those who want to use talkkonnect
as a hobby.

For production I would always recommend you install from scratch using the scripts.
Since there are so many moving parts it is not easy to be able to anticipate what goes wrong cause there are so many dependencies. I have also tried to vendor the libraries so talkkonnect always builds using the same version of the libraries.

What usually happens is talkkonnect fails to build there no hardware damage or permanent damage caused.

In summary if on production dont use the images as they can be pretty old and outdated, please install using the scripts provided on the latest version of hardware and software (OS) you can get your hand on.

Always appreciate all your feedback,

suvir

Pegasus454, 2024-06-04:
Hi Suvir, 

  Thanks for the advice.  I'll gi9ve it another go from a "blank" install of the OS.  I originally tried this, and everything was working with the exception of the GPIO. 

  Last night, when I got home, I powered the unit off and on again and the audio came back, but the GPIO was not working. This was after an "apt update" and "apt upgrade" of the prebuilt Pi4B image. 

  But...  I'll give it another go tonight when I get home.  I'll let you know what I find 8-) 

  Thank you for being patient with me as I sort this out 

Kind Regards,
Peg

talkkonnect, 2024-06-04:
Hi Peg,
use the instructions as found by using the link below to install from scratch.

https://github.com/talkkonnect/talkkonnect?tab=readme-ov-file#installation-instructions-for-raspberry-pi-boards-from-bash-script

suvir

Pegasus454, 2024-06-04:
Hi Suvir, 

  I am really sorry, but I am just not getting anywhere.. I have done the install from a fresh bookworm image that I then upgraded using apt update and apt upgrade.. I then built talkkonnect from source using your script and coped the sample-config  talkkonnect-version2-usb-gpio-example.xml to talkkonnext.xml and started the application.  I get the same result where everything works apart from the GPIO.  

[image]

I know the GPIO is working using the native tools

[image]

Am I just being thick?  (Its ok to say yes!  *LOL*)

 ~Peg

talkkonnect, 2024-06-04:
You are running as root correct? Have to run as root. Also if there really was a change it’s some GPIO library level change. I have not seen this before. I guess will have to try it out this weekend on a raspberry pi 4, time permitting.

Pegasus454, 2024-06-05:
Hi Suvir, 

  It looks like the most recent version of bookworm does not map the GPIO pin's the same way as the previous versions had. I am unclear if this is because they are accommodating something in the newer RP5, they are forcing people to not use SYSFS control any more or something is just borked on my updated system.  

But....

If I add 512 to all GPIO pin numbers, it works.  What I had to do was modify my talkkonnext.xml file with the new port numbers and then modify the code config parser to allow the new range of 512 > X > 544 . 

I know this is not a good way of fixing it, and I may come up with something better shortly, but at least it all seems to work now. 

This is what I get from the kernel:

`root@MHTHermes01:/boot/overlays#  cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 512-569, parent: platform/fe200000.gpio, pinctrl-bcm2711:
 gpio-512 (ID_SDA              )
 gpio-513 (ID_SCL              )
 gpio-514 (GPIO2               )
 gpio-515 (GPIO3               )
 gpio-516 (GPIO4               )
 gpio-517 (GPIO5               |sysfs               ) out hi 
 gpio-518 (GPIO6               |sysfs               ) out lo 
 gpio-519 (GPIO7               )
 gpio-520 (GPIO8               )
 gpio-521 (GPIO9               )
 gpio-522 (GPIO10              )
 gpio-523 (GPIO11              )
 gpio-524 (GPIO12              |sysfs               ) in  lo 
 gpio-525 (GPIO13              |sysfs               ) out lo 
 gpio-526 (GPIO14              )
 gpio-527 (GPIO15              )
 gpio-528 (GPIO16              |sysfs               ) in  hi 
 gpio-529 (GPIO17              |sysfs               ) in  lo 
 gpio-530 (GPIO18              |sysfs               ) in  lo 
 gpio-531 (GPIO19              )
 gpio-532 (GPIO20              )
 gpio-533 (GPIO21              )
 gpio-534 (GPIO22              |sysfs               ) out hi 
 gpio-535 (GPIO23              |sysfs               ) in  lo 
 gpio-536 (GPIO24              |sysfs               ) in  lo 
 gpio-537 (GPIO25              |sysfs               ) in  lo 
 gpio-538 (GPIO26              |sysfs               ) out lo 
 gpio-539 (GPIO27              |sysfs               ) in  lo 
 gpio-540 (RGMII_MDIO          )
 gpio-541 (RGMIO_MDC           )
 gpio-542 (CTS0                )
 gpio-543 (RTS0                )
 gpio-544 (TXD0                )
 gpio-545 (RXD0                )
 gpio-546 (SD1_CLK             )
 gpio-547 (SD1_CMD             )
 gpio-548 (SD1_DATA0           )
 gpio-549 (SD1_DATA1           )
 gpio-550 (SD1_DATA2           )
 gpio-551 (SD1_DATA3           )
 gpio-552 (PWM0_MISO           )
 gpio-553 (PWM1_MOSI           )
 gpio-554 (STATUS_LED_G_CLK    |ACT                 ) out lo 
 gpio-555 (SPIFLASH_CE_N       )
 gpio-556 (SDA0                )
 gpio-557 (SCL0                )
 gpio-558 (RGMII_RXCLK         )
 gpio-559 (RGMII_RXCTL         )
 gpio-560 (RGMII_RXD0          )
 gpio-561 (RGMII_RXD1          )
 gpio-562 (RGMII_RXD2          )
 gpio-563 (RGMII_RXD3          )
 gpio-564 (RGMII_TXCLK         )
 gpio-565 (RGMII_TXCTL         )
 gpio-566 (RGMII_TXD0          )
 gpio-567 (RGMII_TXD1          )
 gpio-568 (RGMII_TXD2          )
 gpio-569 (RGMII_TXD3          )

gpiochip1: GPIOs 570-577, parent: platform/soc:firmware:gpio, raspberrypi-exp-gpio, can sleep:
 gpio-570 (BT_ON               )
 gpio-571 (WL_ON               )
 gpio-572 (PWR_LED_OFF         |PWR                 ) out lo ACTIVE LOW
 gpio-573 (GLOBAL_RESET        )
 gpio-574 (VDD_SD_IO_SEL       |vdd-sd-io           ) out hi 
 gpio-575 (CAM_GPIO            |cam1_regulator      ) out lo 
 gpio-576 (SD_PWR_ON           |regulator-sd-vcc    ) out hi 
 gpio-577 (SD_OC_N             )`

And finally, when I look at the running systems /sys/class/gpio

`root@MHTHermes01:/boot/overlays# ls -la /sys/class/gpio
total 0
drwxrwxr-x  2 root gpio    0 Jun  5 22:31 .
drwxr-xr-x 65 root root    0 Jan  1  1970 ..
--w--w----  1 root gpio 4096 Jun  5 21:53 export
lrwxrwxrwx  1 root root    0 Jun  5 22:30 gpio517 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio517
lrwxrwxrwx  1 root root    0 Jun  5 22:30 gpio518 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio518
lrwxrwxrwx  1 root root    0 Jun  5 22:31 gpio524 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio524
lrwxrwxrwx  1 root root    0 Jun  5 22:31 gpio525 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio525
lrwxrwxrwx  1 root root    0 Jun  5 22:31 gpio528 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio528
lrwxrwxrwx  1 root root    0 Jun  5 22:31 gpio529 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio529
lrwxrwxrwx  1 root root    0 Jun  5 22:31 gpio530 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio530
lrwxrwxrwx  1 root root    0 Jun  5 22:31 gpio534 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio534
lrwxrwxrwx  1 root root    0 Jun  5 22:31 gpio535 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio535
lrwxrwxrwx  1 root root    0 Jun  5 22:31 gpio536 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio536
lrwxrwxrwx  1 root root    0 Jun  5 22:31 gpio537 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio537
lrwxrwxrwx  1 root root    0 Jun  5 22:31 gpio538 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio538
lrwxrwxrwx  1 root root    0 Jun  5 22:31 gpio539 -> ../../devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio539
lrwxrwxrwx  1 root gpio    0 Jun  5 21:53 gpiochip512 -> ../../devices/platform/soc/fe200000.gpio/gpio/gpiochip512
lrwxrwxrwx  1 root gpio    0 Jun  5 21:53 gpiochip570 -> ../../devices/platform/soc/soc:firmware/soc:firmware:gpio/gpio/gpiochip570
--w--w----  1 root gpio 4096 Jun  5 21:53 unexport`

Kind Regards,
Peg

talkkonnect, 2024-06-05:
Some one posted a fix in another forum 

sudo apt-get update; sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel

After that update the software and reboot
sudo apt-get update
sudo apt-get upgrade

Looks like there are some changes in the newer kernel just like you said.  This is affecting the ripoff library that everyone uses for GPIO.

will have to wait and see what is the recommended course of action. Until then should stick to the older kernel for now.

closed 2024-06-07 by talkkonnect