talkkonnect/talkkonnect #130: High bandwidth being sent to server when idle

Pegasus454, 2024-07-02:
Hi Everyone, 

I was wondering if anyone else has seen this or not. I am using Talkkonnect in PTT mode (meaning that I have a GPIO pin assigned to PTT). When I press the PTT, everything works as planned and the clients all hear my audio OK.  However, I have noticed that when the PTT is not pressed, there is a constant 20Mb/s stream being sent to the murmur server. 

It is probably something in my config, but I was wondering if anyone else as seen this?

[image]

talkkonnect, 2024-07-03:
Hi,
have not seen this behavior  of expressive bandwidth use when not transmitting or receiving for additional information please  also monitor the CPU levels using tools like HTOP or TOP and filter to see talkkonnect resource usage so as to see how and when this traffic is generated to recognize if there is a pattern to this behavior. 

best,
suvir

Pegasus454, 2024-07-03:
Hi Suvir, 

Yea.. Its pretty strange but I can tell its definitely talkkonnect sending the data.  Below is a snapshot of what htop looks like

I made sure I went to the console and hit f9 to "stop transmitting" and then did the below htop

[image]

[image]

Is there any detailed debug I can get that may be able to help?

Kind Regards,
Peg

talkkonnect, 2024-07-05:
Pegasus,
I can reproduce the problem here, but looked and looked cant find the source of the problem. Looked and looked until my brain went numb yesterday. It happens as soon as the TCP client dials the server. It looks like a deep rooted problem that will not be easy to find. Will take a lot of time since these days very little time to spare on talkkonect.

Reverted to the orignal gumble and gopus libraries it is still happening. So currently I am out of ideas.

suvir

zrdimetc, 2024-08-06:
When running the latest talKKonnect build in Orange Pi with Murmur server version 1.3.4 installed from a package, (not a compiled version 1.5 or 1.6), talKKonnect client is perfectly quiet and well behaving, There is no unusually high bandwidth utilization and outgoing traffic to the server when standing by. The issue is reproduced only when talKKonnect connects to community server mumble.talkkonnect.com (Murmur compiled version 1.6). When connected to the community server, there is outgoing traffic 20-100 kilobits pers second. This must be a server issue. Possibly related to a bug with some new fearure, a server config or the way server was compiled.

talkkonnect, 2024-08-07:
Zoran,
For me when i last looked it was in the MB/Sec range not KB so it's interesting that different people are getting different results.
Do let us know if you find any underlying root cause.

/suvir

zrdimetc, 2024-08-07:
Bandwidth usage seem to increase with more clients in the channel and traffic is bi-directional. No streaming, all clients are standing by.
[image]
[image]
[image]

zrdimetc, 2024-08-07:
Now, this is my Murmur 1.3.4 with latest talKKonnect build
[image]
[image]
[image]
Network traffic when the clients are idle, is negligible

zrdimetc, 2024-08-07:
Suvir, try to disable channel listener feature to test behavior? In the server? In the client? We already know, that new talKKonnect code with old server, version 1.3.4 works fine. talKKonnect code prior to implementing channel listener with latest server, version 1.6.0 might also work fine? Or just try disabling channel listener in the server and let's observe traffic with nethogs or iftop?

zrdimetc, 2024-08-07:
I tested talKKonnect Ver 2.43.01 Rel 24 Jun 2024 (Latest Release) with Mumble server version 1.6. I dont see any excesive network traffic between idle client and server. They behave as expected. The issue reported, must be related just to talKKonnect Mumble community server in particular. I am unable to recreate the issue with latest talKKonnect and other servers, either versions 1.3.4 ot 1.6. Here is the `nethogs `reading of talKKonnect with Mumble server version 1.6.
[image]

talkkonnect, 2024-09-22:
All,
there is some progress on the high bandwidth font I have found that if we use versions lower than 
https://github.com/mumble-voip/mumble/commit/ea3e7cdce6da6b0f5544d0829871fefd23293938 
this commit of the murmur server the issue is not happening on talkkonnect for excessive bandwidth.

They made some audio buffer fix and that causes havoc when used with talkkonnect. 

Until we find out why this is happening with this ecosystem we will have to refrain from using any mumble server that is newer than November 2023

for now thats all folks until time permits. 

Thanks Zoran for the inspiration to look again today until I got lucky.

suvir kumar

Pegasus454, 2024-09-22:
Oh wow.. Thats really good dective work!1 Thank you for finding this. 

Could you tell me how to download from GIT the version that is still compatible?  I am unsure how to specify a specific commit point to give it a try.

talkkonnect, 2024-09-24:
The version I think you need is e99c7c815d2a24cdd8e068e35f6a420ad2c52c27 thats last  the Github hash for the server version that works with talkkonnect with no bandwidth issue.. 

to get the version that works you should be able to do a 
git checkout e99c7c815d2a24cdd8e068e35f6a420ad2c52c27

have to later see what changes they made, however to fix talkkonnect it would be at the gumble library level which I am not so sure I have enough knowledge to do that.

closed 2024-09-24 by talkkonnect

zrdimetc, 2025-04-20:
Has the bandwidth utllization problem been resolved when using the latest talKKonnect, Mumble server in most recent Debian based Linux distributions? Mumble package 1.3.4 from current Linux disto's and the latest talKKonnect's do not have an issue? High bandwidth when nobody is talking can only be seen between talKKonnect's and Mumble servers version 1.6? Is checking out Mumble server e99c7c815d2a24cdd8e068e35f6a420ad2c52c27 tag still necessary for building that particual version of server or remaining with older versions of Mumble server still necessary to keep bandwidth leak issue under control?