talkkonnect/talkkonnect #132: alcCore.go

FilipStadler, 2024-07-24:
OS: Debian GNU/Linux 12 (bookworm) aarch64  Host: Raspberry Pi 3 Model B Rev 1.2  Kernel: 6.6.31+rpt-rpi-v8

I get a error when I try to run ./update-talkkonnect.sh in /root as root

# github.com/talkkonnect/go-openal/openal
In file included from ../../../../pkg/mod/github.com/talkkonnect/go-openal@v0.0.0-20181013102444-59f5b1d30aff/openal/alcCore.go:24:
./local.h:5:9: fatal error: AL/al.h: No such file or directory
    5 | #include<AL/al.h>
      |         ^~~~~~~~~
compilation terminated.
copying talkkonnect-20240724-135034.xml
cp: kan ikke udføre stat() på '/root/mumble-20240724-135034.pem': Ingen sådan fil eller filkatalog
cp: kan ikke udføre stat() på '/root/talkkonnect-20240724-135034.xml': Ingen sådan fil eller filkatalog
talkkonnect is stopped now restarting talkkonnect
Failed to start talkkonnect.service: Unit talkkonnect.service not found.
=> Finished Updating TalKKonnect
=> Updated talkkonnect binary is in /home/talkkonect/bin
copied old talkkonnect.xml file and replaced in /home/talkkonnect/gocode/src/github.com/talkkonnect/talkkonnect/
Happy talkkonnecting!!

talkkonnect, 2024-07-25:
this is library issue. Install clean os and start clean install with tkbuild.sh
found here https://raw.githubusercontent.com/talkkonnect/talkkonnect/main/scripts/tkbuild.sh

if you upgrade from images there are some problems due to version changes

talkkonnect, 2024-07-29:
Since there is no answer or acknowledgement or feedback I will close this ticket