talkkonnect/talkkonnect #134: Is talkkonnect something for a ham radio remote station?

peterus, 2024-09-21:
I am currently planing a ham radio remote station and still looking for a software to use.
I am using a SA818 module (from [jumbospot SHARI](https://de.aliexpress.com/item/1005003245127518.html?spm=a2g0o.productlist.main.1.3eca3955YKyke4&algo_pvid=f2de4694-8705-4808-b82b-ff6bbbedc557&algo_exp_id=f2de4694-8705-4808-b82b-ff6bbbedc557-0&pdp_npi=4%40dis%21EUR%2159.99%2159.99%21%21%2165.39%2165.39%21%4021038dfc17269160861777874ec951%2112000032104866749%21sea%21AT%212309446643%21X&curPageLogUid=XO0sKT79PSAB&utparam-url=scene%3Asearch%7Cquery_from%3A) project) which is connected via a USB to UART and a external sound card.
I found a Schematic [here](https://wiki.fm-funknetz.de/lib/exe/fetch.php?media=fm-funknetz:technik:shari_construction_manual_v1-07.pdf) which is nearly the same PCB (filter on RF side is missing etc.).

I would like to distribute the voice via a Mumble Server and the Raspberry Pi is connecting to it via a Mumble Client implementation. The SA818 should be configured via the Mumble Chat so I don't need an additional service etc (Mumble Client is taking care).

As your project is already very big and has a lot of features I am wondering is something like this is possible with your code base.
Maybe you can give me some hints where to start. Thank you very much.

talkkonnect, 2024-09-22:
yes  it will work as you want however the sa818 is controlled via http api or mqtt instead of within the chat. you can use it as a remote station from mumble to RF

peterus, 2024-09-23:
thanks for the information. I will take a look on it.

closed 2024-09-23 by peterus