talkkonnect/talkkonnect #135: Circuit Diagram - Fan Rectifier Circuit Diode

SecareLupus, 2024-10-01:
On the Circuit Diagram provided for identifying GPIO configurations, the diode labeled D3 appears to be backwards. In its current configuration, it would allow power supplied via the POWERIN tag to the rest of the circuit, instead of providing stabilized one-way 5v power to the POWERIN (and thus what I assume should be a fan connected to the JST 2510S-2p)

This fact is not relevant to the reason the link was supplied (GPIO wiring), but something I noticed while planning out a breadboarded test device.

talkkonnect, 2024-10-03:
OK thanks for the catch, the intended reason the diode was there was to prevent reverse polarity from the PCB. Noted and will fix the diagram again.

closed 2024-10-07 by talkkonnect

SecareLupus, 2024-10-07:
Oh, the JST is supposed to connect to a supply? I misread your 5v tag as the supply then, and assumed it was a fan for active cooling on the other end of that JST. Thanks for the clarification.