talkkonnect/talkkonnect #136: Multiple LCD/OLED

gilliangoud, 2024-10-16:
Hi!

I've searched through the configs, code and issues and wanted to verify that at this moment in time 
only 1 display is supported, right?

I was thinking of making something like this:
[image]

talkkonnect, 2024-10-18:
The current use case as a desktop transceiver/intercom one display is sufficient. Yes currently only one display is supported.

talkkonnect, 2024-10-19:
If the messages you want to display are simple for example offline/online or speaking etc etc you can use LEDs to do that with a little modification of the code since talkkonnect supports the mcp23017 there are 100s of outputs available.

closed 2024-10-19 by talkkonnect