talkkonnect/talkkonnect #138: Duplicated Confirmation Audio Clips

SecareLupus, 2024-12-02:
I'm building a system which implements talkkonnect as part of its stack, and currently use HTTP Commands to interact with headless devices. I've noticed that the Acknowledgement Tones that play in response to commands are getting played twice in quick succession. Is this likely due to a configuration issue? Is there anything I can do to track down the cause?

talkkonnect, 2024-12-03:
use screen -r if you are using the image to see the logs and do a http call and see if indeed talkkonnect is playing the file twice, and if so you will be able to see the line number of the program causing the issue. Let me know your result so I can have a look at the code if there is indeed a bug.

SecareLupus, 2024-12-03:
I will make time to try this out in the next day or two, thanks for the suggestion!

SecareLupus, 2024-12-09:
I think this may have stopped happening in my testing without figuring out why it was in the first place. Probably some misconfiguration. Thank you for your help.

closed 2024-12-09 by SecareLupus