talkkonnect/talkkonnect #140: Voice target group not working

mdetweiler, 2025-02-06:
I'm trying to reset or remove all voice targets so that I am broadcasting to the whole room again, but can't figure out the correct syntax in the xml section so I attempted to just create a group of users under one voice target id, but that also doesn't seem to work.  

Here is the <voicetargets> section of my config file.   My goal was to select voice target ID #1 and have it revert to sending audio to all of the users in my default channel, but it doesn't seem to work.
I tried this because I couldn't figure out how to just remove the set voice target that I selected using the http api.  Note: ID #6 is just a leftover from the example config that I was hoping to use as an example to create a default ID section...

Could the problem be caused if one of the users is not online?  

      <voicetargets>
        <id value="0">
            <user>xx100</user>
        </id>
        <id value="1">
          <users>
            <user>xx1</user>
            <user>xx2</user>
            <user>xx3</user>
            <user>xx4</user>
            <user>xx5</user>
            <user>XMRD</user>
          </users>
        </id>
        <id value="2">
          <users>
            <user>xx4</user>
          </users>
        </id>
        <id value="3">
          <users>
            <user>xx3</user>
          </users>
        </id>
        <id value="4">
          <users>
            <user>xx2</user>
          </users>
        </id>
        <id value="5">
          <users>
            <user>xx1</user>
          </users>
        </id>
        <id value="6">
          <channels>
            <channel>
              <name>TEST1</name>
              <recursive>true</recursive>
              <links>true</links>
              <group>"all"</group>
            </channel>
          </channels>
        </id>
      </voicetargets>

talkkonnect, 2025-02-07:
Voice target id = 0 should be used to reset back clearing the voice targets in the xml the id should start at 1 not 0. The target 0 is reserved by mumble protocol to clear voice targets in the. Valid targets are from number 1 to number 31

Suvir

mdetweiler, 2025-02-07:
Unfortunately, I'm still having no luck.  I have tried removing the 0th voice target section from the xml and tried it with an empty users section and with an invalid user and with several valid users, but when I submit the http api request to set the voice target to 0, it seems to stay set to the last known choice.  I can select any other voice target the same way and it works so the process seems fine, but it is ignoring my attempts to remove all target and shout to the channel?

      <voicetargets>
        <id value="0">
          <users>
          </users>
        </id>

mdetweiler, 2025-02-07:
Took a look in the code and it looks like maybe the actual method that should clear the voice target is commented out?  I'm not  go programmer though... 

`func (b *Talkkonnect) VoiceTargetUserSet(TargetID uint32, TargetUser string) {
	if len(TargetUser) == 0 && TargetID == 0 {
		TargetUser = b.Client.Self.Name
	}

	vtUser := b.Client.Users.Find(TargetUser)
	if (vtUser != nil) && (TargetID <= 31) {
		vtarget := &gumble.VoiceTarget{}
		vtarget.ID = TargetID
		vtarget.AddUser(vtUser)
		b.Client.VoiceTarget = vtarget
		if TargetID > 0 {
			log.Printf("debug: Added User %v to VT ID %v\n", TargetUser, TargetID)
			b.sevenSegment("voicetarget", strconv.Itoa(int(TargetID)))
			GPIOOutPin("voicetarget", "on")
		} else {
			//b.VoiceTarget.Clear()
			GPIOOutPin("voicetarget", "off")
			log.Println("debug: Cleared Voice Targets")
			b.sevenSegment("voicetarget", strconv.Itoa(int(TargetID)))
		}
		b.Client.Send(vtarget)
	} else {
		log.Printf("error: Cannot Add User %v to VT ID %v\n", TargetUser, TargetID)
	}`

talkkonnect, 2025-02-08:
OK Good Catch I must have been testing something and forgot to remove the comment in the code
//b.VoiceTarget.Clear()
please remove the // 
and recompile
with /root/compile if you are using the image

suvir

mdetweiler, 2025-02-08:
That fixed it!

I does seem to require a voice target id 0 section to work though.  

      <voicetargets>
        <id value="0">
          <users>
            <user>xx1</user>
          </users>
        </id>
        <id value="1">

talkkonnect, 2025-02-09:
Dear mdetweiler,
thank for your valuable feedback for the community, I have fixed the code in the latest release. 
I will close this ticket now.

suvir

closed 2025-02-09 by talkkonnect