talkkonnect/talkkonnect #141: raspberry pi 5 GPIO issue

sivascu, 2025-02-09:
Hello, 
It seems the raspberry pi 5 users a different chip for GPIO (found some info here: https://forums.raspberrypi.com/viewtopic.php?t=359829 ).
Could you recommend a fix for this ?

talkkonnect, 2025-02-10:
Hi,
Currently there is no support for raspberry pi 5 as the libraries for GPIO have completely changed, also I currently do now have a Pi5 Board to test it on. Talkkonnect will work from raspberry pi 2 till 4 all variations and also on orange pi zero (some versions).

Another factor is Raspberry Pi 5 is overkill (very high specs) not needed by talkkonnect application.

Not sure when I will get around to testing and developing for Pi5.

Currently there is no fix but perhaps you could change the gpio library try.

sivascu, 2025-02-10:
Thanks for the response. 
I can confirm it's working on PI5 using the installation steps from the readme as long as you are not using the gpio inputs. 
For now I'm using headsets with a mute button (and I've set streamonstart to true), but I've also tested the API control from a stream deck (and the companion app) and that also works.
I'll look at how the gpio library changes and if I find any good solutions will get back to you.

closed 2025-02-10 by sivascu